serialport/node-serialport #939: DSRDTR not working

neilger, 2016-09-23:
SerialPort version: 4.0.1
NodeJS Version: 4.2.6
Operating System and Hardware: GalliumOS 2.0 (16.04.1)
Are you using Electron/Atom/NwJS?: no

### Summary of Problem

RTSCTS flow control is working for a machine interface, but not DSRDTR

### Expected Behavior

### Actual Behavior

### Steps and Code to Reproduce the Issue

From a browser client for rapid-prototyping workflows, e.g.:

   http://mods.cba.mit.edu/?program=programs/machines/Roland/mill/MDX-20/PCB

I open a connection to a machine serial server:

   http://mods.cba.mit.edu/js/serialserver.js

This opens the serial port with the appropriate flow control:

      if ((flow == 'none') || (flow == 'dsrdtr'))
         port = new SerialPort(device,{baudRate:baud})
      else if (flow == 'rtscts')
         port = new SerialPort(device,{baudRate:baud,rtscts:true})
      port.on('open',function() {
         ws.send('serial port opened')
         if (flow == 'dsrdtr') {
            port.set({dsr:true,dtr:true})
            port.set({rts:false,cts:false})
            }
         })

Characters are then written out one at a time to be able to follow and interrupt progress:

   port.write(msg.contents[count],function(){
      port.drain(function(err){
         if (err)
            ws.send('error '+err.message)
         else {
            ws.send((count+1)+'/'+msg.contents.length)
            count += 1
            if (count < msg.contents.length)
               write_char()
            else {
               console.log('done')
               ws.send('done')
               }
            }
         })
      })

With an RTSCTS machine interface this properly starts and stops based on the machine's progress. But with a DSRDTR machine interface the output never pauses.

In a pyserial version:

   http://fabmodules.org/mod_server/mod_serial.py

this is handled by manually checking is DSR line:

   if (flow == "dsrdtr"):
      while (s.getDSR() != True):
         time.sleep(0.001)

Is port.set imposing DSR flow control, or does it just set the lines? If the latter, how are they read as well as written?

reconbot, 2016-10-05:
> Is port.set imposing DSR flow control, or does it just set the lines?

If I understand you correctly, we just set the lines. Currently we can't read the DSR lines.

reconbot, 2016-10-05:
We talk about this over at #291

closed 2016-10-05 by reconbot

neilger, 2016-10-05:
If you can set DTR but not read DSR, there's no way to implement DSRDTR
flow control, which is required in this case for a machine interface.

If you're already using the tty ioctls, that input should be easy to add.

Neil

On Wed, Oct 5, 2016 at 12:51 PM, Francis Gulotta <notifications@github.com>
wrote:

> Closed #939
> <https://github.com/EmergingTechnologyAdvisors/node-serialport/issues/939>
> .
>
On Wed, Oct 5, 2016 at 12:49 PM, Francis Gulotta <notifications@github.com>
 wrote:

> ​​
> Is port.set imposing DSR flow control, or does it just set the lines?
>
> If I understand you correctly, we just set the lines. Currently we can't
> read the DSR lines.
>