serialport/node-serialport #941: Problem with emitting events

henrysachs, 2016-09-25:
SerialPort version: 4.0.1
NodeJS Version: 6.6.0
Operating System and Hardware: raspberry pi and arduino mega connected
Are you using Electron/Atom/NwJS?:

(Run `node -v && npm -v && python --version` to get version numbers)

### Summary of Problem
the arduino just repeats the send message
### Expected Behavior
execution of the send commands
### Actual Behavior
repeats the send message 
### Steps and Code to Reproduce the Issue
Hi Guys i'm using the newest serialport library and node js. I also use socket.io in combination with it. So now i want to write commands like move to my arduino which than drives. But with the on data event i just get the same message as send back and the arduino does nothing. Here is my code:

`var http = require('http');

var fs = require('fs');
Serialport = require('serialport');
port = '/dev/ttyACM0';
// Loading the file index.html displayed to the client

var server = http.createServer(function(req, res) {

    fs.readFile('./index.html', 'utf-8', function(error, content) {

        res.writeHead(200, {"Content-Type": "text/html"});

        res.end(content);

    });

});

serial = new Serialport(port, {baudrate: 115200});

port.on('open', function() {
  port.write('main screen turn on', function(err) {
    if (err) {
      return console.log('Error on write: ', err.message);
    }
    console.log('message written');
  });
});
 
// open errors will be emitted as an error event 
port.on('error', function(err) {
  console.log('Error: ', err.message);
})

//Funktion um daten vom Arduino zu empfangen

serial.on('data', function(chunk){
readData  =chunk.toString();
if(readData.indexOf('\n')){
io.sockets.emit('message',"Arduino sends: "+ readData);
} });

// socket.io initialisieren

var io = require('socket.io').listen(server);

io.sockets.on('connection', function (socket) {

    // WillkommensgruÃŸ
    console.log('Client verbunden');
    socket.emit('message', 'Willkommen');

    // Falls multiuser irgendwann realisiert sein sollte
    // wird eine nachricht ausgegeben das ein neuer user beigetreten ist

    socket.broadcast.emit('message', 'neuer Client');

    // Wenn das message event ausgelÃ¶st wird wird es in der Konsole geloggt

    socket.on('message', function (message){
      console.log(message);
        });
    socket.on('move', function(message){
        // move
      var message2 = new Buffer(message);
       console.log(message2.toString());
       return serial.write(message2);
	     });
    socket.on('start', function(message){
      //motor 1
      var message2 = new Buffer("motor 1");
      console.log(message2.toString());
	serial.write(message2, function(err) {
    	if (err) {
      return console.log('Error on write: ', err.message);
    }
    console.log('message written');
  	});

      });
      socket.on('stop', function(message){
          //stop
      var message2 = new Buffer("stop");
      console.log(message2.toString());
      return serial.write(message2);
                });
      socket.on('poke', function(message){
                    //stop
      var message2 = new Buffer("ping");
      console.log(message2.toString());
      return serial.write(message2);
                          });
    socket.on('sensoron', function(message){
                                        //stop
    var message2 = new Buffer("sensors 1");
    console.log(message2.toString());
    return serial.write(message2);
                          });
    socket.on('sensoroff', function(message){
                                                            //stop
    var message2 = new Buffer("sensors 0");
    console.log(message2.toString());
    return serial.write(message2);
                          });

});

server.listen(8080);
`

And my Website:

`
<!DOCTYPE html>
<html>
    <head>
        <meta charset="utf-8" />
        <title>Socket.io</title>
	<script src="https://ajax.googleapis.com/ajax/libs/jquery/1.12.4/jquery.min.js"></script>
	<script src="/socket.io/socket.io.js"></script>

    </head>

    <body>
        <h1>Communicating with socket.io!</h1>

	<table>
	  <tr>
		<td>Geschwindigkeit</td>
		<td>Neigung</td>
	  </tr>
	  <tr>
		<td><input type="text" name="geschwindigkeit" id="v"></td>
		<td><input type="text" name="Neigung" id="acc"></td>
	  </tr>
	</table>

<p> </p>
  <table>
  <tr>
    <td><input type="button" value="Move" id="move" /></td>
    <td><input type="button" value="Start" id="start" /></td>
    <td><input type="button" value="test" id="test" /></td>
  </tr>
  <tr>
    <td><input type="button" value="Stop" id="stop" /></td>
    <td><input type="button" value="Ping" id="ping" /></td>
    <td> <p> </p> </td>
  </tr>
  </table>

        <script>
	       var socket = io.connect();

            // A dialog box is displayed when the server sends us a "message"
            socket.on('message', function(message) {
                alert('Nachricht: ' + message);
            })
	 // When the button is clicked, a "message" is sent to the server
            $('#start').click(function () {
                socket.emit('start', 'start');
            })

 		         $('#test').click(function () {
                socket.emit('message', 'test');
            })

            $('#move').click(function () {
               	  var geschw = document.getElementById("v").value;
		              var neig = document.getElementById("acc").value;
		              socket.emit('move', 'move ' + geschw + ' ' + neig);
            })

            $('#stop').click(function () {
                socket.emit('stop', 'Car stopped');
            })

            $('#ping').click(function () {
                socket.emit('poke', 'Hallo');
            })

        </script>
        <p></p>
        "[image]
    </body>
</html>
`

reconbot, 2016-09-28:
Off the top of my head it looks like you need to specify the `baudRate` with a capital `R` it won't work lowercase.

henrysachs, 2016-09-28:
Hi reconbot,

i think i already found my error. I got an arduino sketch provided by some colleagues and they use the cmd.h library but it only listens on commands send by a terminal. So i just have to write my own sketch thanks for your help anyway. :)

closed 2016-10-05 by reconbot