serialport/node-serialport #943: Crashes electron

spiraljacobs, 2016-09-27:
SerialPort version: 4.0.1
NodeJS Version: v6.2.2
Operating System and Hardware:Windows 10 Anniversary Edition 64bit
Are you using Electron/Atom/NwJS?:Electron 1.1.2

### Summary of Problem
Putting:
 let SerialPort = require('serialport');
at the top of a js renderer file (ie. index.js) will cause electron to crash very quickly, without an exception showing up.

If I instantiate in a method, say, a start button method, then it will crash if I attempt to receive data:
 let SerialPort = require('serialport');
let port = new SerialPort('COM4');
      port.on('error', showError);

      port.on('data', ()=> {
        console.log(data.toString());
      })

      port.on('open', ()=> {
          console.log("port open");
          port.write('<OK?>', function(err) {
            if (err) {
              return console.log('Error on write: ', err.message);
            }
          });
      });

### Steps and Code to Reproduce the Issue
Electron "1.1.2"
used "node-pre-gyp install --fallback-to-build --runtime=electron --target=1.1.2 --target_arch=ia32 --directory=node_modules/serialport/ --update-binary --dist-url=https://atom.io/download/atom-shell"
to rebuild serialport.

reconbot, 2016-09-27:
electron doesnt support binary modules very well, we get this question a
lot and almost every time they have changed something about how to make it
work. With serialport 5x we'll be able to make this a lot easier. With both
a chrome serialport backbend and electron prebuilt binaries.

spiraljacobs, 2016-09-27:
I would just love a valid piece of code from anyone that can demonstrate that serialport works in Electron, as of right now, I haven't been able to get it to work with any kind of reliability - it either behaves bizarrely, returning true for errors and won't open ports, or it will just crash Electron completely.  If there's anyone out there with a github with a basic Electron app with serialport working in it, please, post it here.  I'd rather use serialport than create my own node.js addon for serialport communication.

@reconbot  When is the projected initial release of 5x w/ electron support?

reconbot, 2016-09-27:
We have an electron megathread open "tracking electron support" full of
people who have gotten it to work.

spiraljacobs, 2016-09-27:
@reconbot I know, I've been trying those things, but as of yet, none of them have worked in a way that allows serialport to be used without crashing Electron for me.  A simple github w/ code and scripts to get it running would go a long way.

reconbot, 2016-09-27:
Ask on that thread the right people will hear you.

spiraljacobs, 2016-09-27:
@reconbot thanks for the help!  Any rough timelines on 5.0 w/ electron working?

reconbot, 2016-09-27:
A few weeks

---
Francis Gulotta
wizard@roborooter.com

reconbot, 2016-09-27:
But I want to stress 4x works with electron, lots of people are doing it,
but plenty as frustrated as you, I've put out a call to help write some
docs. https://twitter.com/reconbot/status/780765013518745600

---
Francis Gulotta
wizard@roborooter.com

spiraljacobs, 2016-09-27:
@reconbot A few weeks would be totally awesome!  I switched gears and I've been working on a few other things to aleviate the frustration with trying to get this to work, but I'll head back in there to try to get 4.x working sometime this week.  
Thanks again, and hopefully someone will help put up a basic github with a barebones electron > 1.1.x that works with serialport, with scripts to get it set up properly.

reconbot, 2016-09-28:
I'm going to close this in favor of #538, I think a boilerplate would be a neat idea. People have posted step by step instructions in that thread and we'll see how the docs shake out.

closed 2016-09-28 by reconbot