serialport/node-serialport #947: Easy to fix: Uncaught exception when `_afterRead` is called and SerialPort is paused

bminer, 2016-09-30:
SerialPort version: 4.0.1
NodeJS Version: 6.7.0
Operating System and Hardware: Linux Mint 17.3
Are you using Electron/Atom/NwJS?: No

### Summary of Problem
Uncaught exception on line 287 of serialport.js.  This bug should be an easy fix.

Snippet below:
```javascript
       // do not emit events if the stream is paused
        if (this.paused) {
          this.buffer = Buffer.concat([this.buffer, b]);
          return;
        }
```

### Expected Behavior
Should check to ensure `this.buffer` is not `null` to avoid the uncaught exception.  `this.buffer` is explicitly set to `null` in `SerialPort.prototype.resume`, but it is not set to an empty Buffer in `SerialPort.prototype.pause`.

### Actual Behavior
An uncaught exception occurs

### Steps and Code to Reproduce the Issue
Not sure how to reproduce, but this bug occurs when the SerialPort is paused and resumed a few times and `_afterRead` is called while it is paused.

bminer, 2016-09-30:
Maybe change to the following:

```javascript
        // do not emit events if the stream is paused
        if (this.paused) {
          if(this.buffer)
            this.buffer = Buffer.concat([this.buffer, b]);
          else
            this.buffer = b;
          return;
        }
```

reconbot, 2016-09-30:
This is already fixed in master.

reconbot, 2016-09-30:
=)

bminer, 2016-09-30:
@reconbot - Awesome, thanks. Will there be any more 4.x updates, or should I just update to 5.x?

closed 2016-09-30 by bminer

bminer, 2016-10-04:
Duplicate of #945 and #925.

reconbot, 2016-10-05:
I've released [Serialport@4.0.3](https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/tag/4.0.3) which includes this fix. 5x is still a few weeks out so I backported it.