serialport/node-serialport #948: Errors during Serialport installation

OwenLearnsToCode, 2016-10-01:
SerialPort version: 4.0.1
NodeJS Version: 4.5.0
Python Version: 2.7.12
Node-pre-gyp version: 0.6.30
Operating System and Hardware: Snappy Ubuntu Core on Raspberry Pi 2
Are you using Electron/Atom/NwJS?: No

(Run `node -v && npm -v && python --version` to get version numbers)

### Summary of Problem
Serialport module does not install correctly. Possibly due to node-pre-gyp errors.

### Expected Behavior
Expected successful installation after running `sudo npm install serialport` without errors.

### Actual Behavior
Two error messages causing missing files in the serialport module.

### Steps and Code to Reproduce the Issue
```
ubuntu@ubuntu:~$ sudo npm install serialport
-
> serialport@4.0.1 install /home/ubuntu/node_modules/serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download: https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.1/serialport-v4.0.1-node-v46-linux-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.1 and node@4.5.0 (node-v46 ABI) (falling back to source compile with node-gyp)
make: Entering directory '/home/ubuntu/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
  CXX(target) Release/obj.target/serialport/src/serialport_unix.o
  CXX(target) Release/obj.target/serialport/src/serialport_poller.o
  SOLINK_MODULE(target) Release/obj.target/serialport.node
  COPY Release/serialport.node
make: Leaving directory '/home/ubuntu/node_modules/serialport/build'
serialport@4.0.1 node_modules/serialport
├── bindings@1.2.1
├── es6-promise@3.3.1
├── commander@2.9.0 (graceful-readlink@1.0.1)
├── nan@2.4.0
├── debug@2.2.0 (ms@0.7.1)
└── object.assign@4.0.4 (function-bind@1.1.0, object-keys@1.0.11, define-properties@1.1.2)
```

When attempt to use any node-pre-gyp command: 

```
ubuntu@ubuntu:~$ node-pre-gyp info
node-pre-gyp info it worked if it ends with ok
node-pre-gyp info using node-pre-gyp@0.6.30
node-pre-gyp info using node@4.5.0 | linux | arm
node-pre-gyp ERR! UNCAUGHT EXCEPTION
node-pre-gyp ERR! stack Error: Cannot find module 'aws-sdk'
node-pre-gyp ERR! stack at Function.Module._resolveFilename (module.js:325:15)
node-pre-gyp ERR! stack at Function.Module._load (module.js:276:25)
node-pre-gyp ERR! stack at Module.require (module.js:353:17)
node-pre-gyp ERR! stack at require (internal/module.js:12:17)
node-pre-gyp ERR! stack at unpublish (/usr/local/lib/node_modules/node-pre-gyp/lib/info.js:14:15)
node-pre-gyp ERR! stack at Object.self.commands.(anonymous function) as info
node-pre-gyp ERR! stack at run (/usr/local/lib/node_modules/node-pre-gyp/bin/node-pre-gyp:79:30)
node-pre-gyp ERR! stack at Object. (/usr/local/lib/node_modules/node-pre-gyp/bin/node-pre-gyp:131:1)
node-pre-gyp ERR! stack at Module._compile (module.js:409:26)
node-pre-gyp ERR! stack at Object.Module._extensions..js (module.js:416:10)
node-pre-gyp ERR! System Linux 4.4.0-1023-raspi2
node-pre-gyp ERR! command "/usr/local/bin/node" "/usr/local/bin/node-pre-gyp" "info"
node-pre-gyp ERR! cwd /home/ubuntu
node-pre-gyp ERR! node -v v4.5.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.30
node-pre-gyp ERR! This is a bug in node-pre-gyp.
node-pre-gyp ERR! Try to update node-pre-gyp and file an issue if it does not help:
node-pre-gyp ERR! https://github.com/mapbox/node-pre-gyp/issues
```

fivdi, 2016-10-02:
Although `sudo npm install serialport` does report the following errors the errors can be ignored:

```
node-pre-gyp ERR! Tried to download: https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.1/serialport-v4.0.1-node-v46-linux-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.1 and node@4.5.0 (node-v46 ABI) (falling back to source compile with node-gyp)
```

There are 32 and 64-bit pre-built serialport binaries for Linux, Windows, and OSX on Intel processors. However, there are no pre-built binaries for ARM processors and your system appears to have an ARM processor. On systems with an ARM processor the above error messages will be printed when installing. The install process then falls back to compiling serialport. This compilation process worked successfully on your system and serialport was installed successfully.

reconbot, 2016-10-04:
Yeah I agree, this looks like it worked.

closed 2016-10-04 by reconbot