serialport/node-serialport #949: Not able to use serialport with Electron (Error in path.js)

akshatgoel, 2016-10-01:
SerialPort version: 4.0.1
NodeJS Version: v6.7.0
Operating System and Hardware: OS X. 
Are you using Electron/Atom/NwJS?: 
Electron - "electron": "^1.3.4",
    "electron-devtools-installer": "^2.0.1",
    "electron-packager": "^7.7.0",
    "electron-rebuild": "^1.2.1",

Using electron boilerplate from https://github.com/chentsulin/electron-react-boilerplate

### Summary of Problem
Unable to require serialport in nodeJS when using webpack and electron. Using boilerplate code from above repository.

### Expected Behavior
No error on requiring serialport

### Actual Behavior
Requiring serialport using require('serialport') results in 'Path must be a string. Received undefined' error.

### Steps and Code to Reproduce the Issue
1. Clone above repository.
2. Run npm install
3. Run npm install  serialport --save
4. Run npm run dev

See the error.

There were other errors as well which I resolved using workarounds.
1. No loader for ReadMe.md - Used html-loader
2. Module vertx not defined (es6-promises.js) - Added vertx in plugins to ignore in webpack

reconbot, 2016-10-01:
You have to not bundle any binary modules with webpack, including
serialport. You can set an array of ignored packages it in your config.

closed 2016-10-11 by reconbot

Heartnett, 2017-07-31:
@reconbot I am also having the same issue, can you provide an example of your explanation?

hanfengcan, 2017-07-31:
Are you rebuild the SerialPort ?  Use node-gyp rebuild before use

Heartnett, 2017-07-31:
I used the rebuild-all command, but it still does not work with electron.

hanfengcan, 2017-07-31:
`"build-dll": "cross-env NODE_ENV=development node --trace-warnings -r babel-register ./node_modules/webpack/bin/webpack --config webpack.config.renderer.dev.dll.js --colors",`

this command is not rebuild serialport

serialport is a Native Node Module, look at here https://electron.atom.io/docs/tutorial/using-native-node-modules/

I use this command rebuild the module
`node-pre-gyp rebuild --target=1.6.2 --arch=x64 --target_arch=x64 --dist-url=https://atom.io/download/electron`

Heartnett, 2017-07-31:
Thanks, I will try it out.

Heartnett, 2017-07-31:
I've tried the rebuild command you recommended, but it's still not working.
I'm getting the following output:

> D:\dev\node\usb-test\node_modules\serialport> node-pre-gyp rebuild --target=1.6.2 --arch=x64 --target_arch=x64 --dist-url=https://atom.io/download/electron
node-pre-gyp info it worked if it ends with ok
node-pre-gyp info using node-pre-gyp@0.6.36
node-pre-gyp info using node@8.2.1 | win32 | x64
[serialport] Removing "D:\dev\node\usb-test\node_modules\serialport\build\Release"
gyp info it worked if it ends with ok
gyp info using node-gyp@3.6.2
gyp info using node@8.2.1 | win32 | x64
gyp info ok
gyp info it worked if it ends with ok
gyp info using node-gyp@3.6.2
gyp info using node@8.2.1 | win32 | x64
gyp info spawn C:\Users\USER\.windows-build-tools\python27\python.EXE
gyp info spawn args [ 'C:\\Users\\USER\\AppData\\Roaming\\npm\\node_modules\\node-gyp\\gyp\\gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'msvs',
gyp info spawn args   '-G',
gyp info spawn args   'msvs_version=auto',
gyp info spawn args   '-I',
gyp info spawn args   'D:\\dev\\node\\usb-test\\node_modules\\serialport\\build\\config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\USER\\AppData\\Roaming\\npm\\node_modules\\node-gyp\\addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\USER\\.node-gyp\\iojs-1.6.2\\common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=C:\\Users\\USER\\.node-gyp\\iojs-1.6.2',
gyp info spawn args   '-Dnode_gyp_dir=C:\\Users\\USER\\AppData\\Roaming\\npm\\node_modules\\node-gyp',
gyp info spawn args   '-Dnode_lib_file=C:\\Users\\USER\\.node-gyp\\iojs-1.6.2\\<(target_arch)\\iojs.lib',
gyp info spawn args   '-Dmodule_root_dir=D:\\dev\\node\\usb-test\\node_modules\\serialport',
gyp info spawn args   '-Dnode_engine=v8',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'D:\\dev\\node\\usb-test\\node_modules\\serialport\\build',
gyp info spawn args   '-Goutput_dir=.' ]
gyp info ok
gyp info it worked if it ends with ok
gyp info using node-gyp@3.6.2
gyp info using node@8.2.1 | win32 | x64
gyp info spawn C:\Program Files (x86)\MSBuild\14.0\bin\msbuild.exe
gyp info spawn args [ 'build/binding.sln',
gyp info spawn args   '/clp:Verbosity=minimal',
gyp info spawn args   '/nologo',
gyp info spawn args   '/p:Configuration=Release;Platform=x64' ]
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  serialport.cpp
  serialport_win.cpp
  win_delay_load_hook.cc
..\src\serialport_win.cpp(158): warning C4311: 'type cast': pointer truncation from 'HANDLE' to 'int' [D:\dev\node\usb-test\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(158): warning C4302: 'type cast': truncation from 'HANDLE' to 'int' [D:\dev\node\usb-test\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(168): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [D:\dev\node\usb-test\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(175): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [D:\dev\node\usb-test\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(185): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [D:\dev\node\usb-test\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(187): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [D:\dev\node\usb-test\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(191): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [D:\dev\node\usb-test\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(193): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [D:\dev\node\usb-test\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(197): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [D:\dev\node\usb-test\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(199): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [D:\dev\node\usb-test\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(204): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [D:\dev\node\usb-test\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(216): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [D:\dev\node\usb-test\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(226): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [D:\dev\node\usb-test\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(297): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [D:\dev\node\usb-test\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(311): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [D:\dev\node\usb-test\node_modules\serialport\build\serialport.vcxproj]
[... cut ...]
node-pre-gyp ERR! node -v v8.2.1
node-pre-gyp ERR! node-pre-gyp -v v0.6.36
node-pre-gyp ERR! not ok
Failed to execute 'C:\Program Files\nodejs\node.exe C:\Users\USER\AppData\Roaming\npm\node_modules\node-gyp\bin\node-gyp.js build --target=1.6.2 --arch=x64 --target_arch=x64 --dist-url=https://atom.
io/download/electron --module=D:\dev\node\usb-test\node_modules\serialport\build\Release\serialport.node --module_name=serialport --module_path=D:\dev\node\usb-test\node_modules\serialport\build\Releas
e' (1)

hanfengcan, 2017-07-31:
`gyp ERR! stack Error: C:\Program Files (x86)\MSBuild\14.0\bin\msbuild.exe failed with exit code: 1`

make sure your node-gyp is ok 

look here: https://github.com/nodejs/node-gyp#on-windows

make sure you can rebuild in node environment

and then rebuild in electron environment

Heartnett, 2017-07-31:
OK I got it to build on the electron environment. Part 2 to the issue - how do I tell webpack to ignore serialport?

hanfengcan, 2017-08-01:
I don't ignore.

It will throw some err when u update you code.

I use electron + vue + serialport,cli template is here https://github.com/SimulatedGREG/electron-vue