serialport/node-serialport #950: Cannot use serialport on RasPI 2 when run in sudo

simonbaev, 2016-10-02:
SerialPort version: 4.0.1 
NodeJS Version:  4.6.0, npm 2.15.9, Python 2.7.9
Operating System and Hardware: RasPi 2 with Raspbian Jessie (installed on 09/16)
Are you using Electron/Atom/NwJS?: no

(Run `node -v && npm -v && python --version` to get version numbers)

### Summary of Problem
I have both local and global installations of serialport. Application works as expected when started under normal account, but when it is started with sudo (I need it to add raspi-io functionality into the app) it throws the following exception: 
`
module.js:327
    throw err;
    ^

Error: Cannot find module '/home/pi/pitron/T1.js'
    at Function.Module._resolveFilename (module.js:325:15)
    at Function.Module._load (module.js:276:25)
    at Function.Module.runMain (module.js:441:10)
    at startup (node.js:139:18)
    at node.js:974:3
`

I am running T1.js without any arguments 

### Expected Behavior
app when started under sudo runs in the same way as without sudo

### Actual Behavior
app when started under sudo throws exception

### Steps and Code to Reproduce the Issue
code attached
[T1.js.zip](https://github.com/EmergingTechnologyAdvisors/node-serialport/files/505188/T1.js.zip)

fivdi, 2016-10-03:
The error message indicates that the file `/home/pi/pitron/T1.js` doesn't exist. Are you 100% sure that it's there? Check with `ls /home/pi/pitron/T1.js`. Note that this issue isn't related to serialport.

simonbaev, 2016-10-03:
I apologize for the dummy issue, please disregard it. I found what was the
problem, and it is not serialport. I was mounting my desktop (Mac)
directory on RasPi 2 (via SSHFS) to have convenient editing experience. I
then sym-linked the file from mounted directory into working directory that
has node_modules (with serialport). I was able to run the application in
non-sudo mode by using that symlink, but it didn't work in sudo (I don't
know why).

Now I reversed the mounting scheme and keep the original JS file on Pi, and
mount project directory from Pi to desktop for editing. With that fix I can
run the app in both sudo and non-sudo mode.

closed 2016-10-04 by reconbot