serialport/node-serialport #953: Writing data gets slower after successive writes

aliask, 2016-10-06:
SerialPort version: 4.0.1
NodeJS Version: 6.7.0
Operating System and Hardware: Windows 10, Rasbian Jessie
Are you using Electron/Atom/NwJS?: No

### Summary of Problem

Using the following code:
```
  var startTime = new Date().getTime();
  writeAndDrain(Buffer.concat([frameHeader, frame]), function(err) {

    var stopTime = new Date().getTime();
    console.log('Took', (stopTime - startTime), 'ms to send');
```

(writeAndDrain taken from docs)

I see the following behaviour for fixed sized payload (387 bytes):
Took 2 ms to send
Took 0 ms to send
Took 1 ms to send
Took 1 ms to send
Took 14 ms to send
Took 33 ms to send
Took 53 ms to send
Took 58 ms to send
Took 58 ms to send
[ Stabilizes at 58-60ms ]

This is causing me issues because I am trying to send a serial message more often than 60ms. I am seeing this same behaviour on Windows 10 and on Linux (Rasbperry Pi ARM6 and ARM7).

How should I go about debugging this performance issue?

aliask, 2016-10-08:
This turned out to be caused by an issue on the serial endpoint, not an issue with anything in node-serialport!

closed 2016-10-08 by aliask

reconbot, 2016-10-08:
Great!