serialport/node-serialport #957: SerialPort.parsers "readline" should be camel case, or documentation should be updated

w4ilun, 2016-10-08:
SerialPort version: 4.0.3
NodeJS Version: 6.3.1
Operating System and Hardware: OS X
Are you using Electron/Atom/NwJS?: No

### Summary of Problem
SerialPort.parsers "readline" should be camel case, or documentation should be updated
### Expected Behavior
ReadLine
### Actual Behavior
readline
### Steps and Code to Reproduce the Issue
```
console.log(SerialPort.parsers.ReadLine);
```

reconbot, 2016-10-12:
You're looking at the docs for master (5x-beta) when you're using 4.x. they match if you look at the right branch. 

- 4x Code https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/4.0.3/lib/parsers.js#L12
- 4x Docs https://github.com/EmergingTechnologyAdvisors/node-serialport/tree/4.0.3#parsers

However, I think the parser in 5x should be `Readline` in master not `ReadLine` which fits both the GNU and the nodejs readline conventions.

closed 2016-10-12 by reconbot

w4ilun, 2016-10-12:
Thanks @reconbot !

PR #963 (2016-10-13): rename ReadLineParser to ReadlineParser
matches gnu and nodejs conventions for `readline`

closes #957