serialport/node-serialport #958: "data"-event is not triggert by incomming data

alex1891, 2016-10-09:
SerialPort version: 4.0.3
NodeJS Version: 6.7.0
Python Version: 2.7.9
Operating System and Hardware: Raspbian Jessy (Raspberry Pi)
Are you using Electron/Atom/NwJS?:

(Run `node -v && npm -v && python --version` to get version numbers)

### Summary of Problem
Not possible to receive data from my serial-port. Writing to the port works fine.

### Expected Behavior
It should be possible to receive data.

### Actual Behavior
"data"-Event is nerver triggert (with Minicom I see that data is received). Other programs can use the serialport without any problems. So the settings should be fine. The device on the serialport is an aduino with the CUL-FW (https://sourceforge.net/p/culfw/code/HEAD/tree/) flashed on it (Baudrate: 38400, 8N1).  

### Steps and Code to Reproduce the Issue

Here is my code snipped:

var SerialPort = require('serialport');

var myPort = new SerialPort('/dev/ttyUSB0', {
	baudrate: 38400,
	parser: SerialPort.parsers.raw
});

myPort.on('open',onOpen);
myPort.on('data',onData);
myPort.on('error', onError);
myPort.on('disconnect', onDisconnect);

function onOpen(){
    console.log('Connection Open');
}

function onData(data){
    console.log('Data: ' + data);
}

function onError(){
    console.log('Something went wrong');
}

function onDisconnect(){
    console.log('Disconnected');
}`

reconbot, 2016-10-11:
It looks like the settings for connection should be ok. Does the device always give data or do you need to write to it first?

alex1891, 2016-10-11:
it gives data after I wrote someting to the port. For example "V01\n" returns the version of the firmware on the aduino. But when I worte "X01\n" one time, it always returns data. I tried it with a python lib now and it works fine. But I want use node.js :(

reconbot, 2016-10-12:
Can you share your working Python proof of concept and a similar nodejs
example?

alex1891, 2016-10-12:
Sure!
Ich added one line to my code above **(bolt-text)**.

Then for my test with pyton i used the file that is attached and startet this serial to tcp bridge with the same arguments (python serialToNetwork.py /dev/ttyUSB0 38400 -P 5555) and than i used  node.js to make a tcp connection to my bridge:

var net = require('net');

var client = new net.Socket();
client.connect(5555, '127.0.0.1', function() {
	console.log('Connected');
	client.write('X01\n');
});

client.on('data', function(data) {
	console.log('Received: ' + data);
});

client.on('close', function() {
	console.log('Connection closed');
});

And it works fine. I receive my data.

reconbot, 2016-10-12:
Great!

closed 2016-10-12 by reconbot