serialport/node-serialport #964: 'version mismatch' error on Linux Ubuntu

Arrow7000, 2016-10-13:
SerialPort version: 4.0.3
NodeJS Version: 6.8.0
Operating System and Hardware: Linux Ubuntu 16.04 LTS
Python version: 2.7.12
Are you using Electron/Atom/NwJS?: No

### Summary of Problem
Serialport throws a 'version mismatch' error when trying to run it in Node v6.8.0. 

I can get SerialPort to work in my script in Node v4.x but any version of Node higher than that throws the same version mismatch error.

On installing it SerialPort does seem to build the module but it still doesn't work.

This is the error message I get when trying to run my script:

```
/home/autocular/Documents/Tracker-3/device/node_modules/bindings/bindings.js:83
        throw e
        ^

Error: Module version mismatch. Expected 48, got 46.
    at Error (native)
    at Object.Module._extensions..node (module.js:597:18)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.require (module.js:497:17)
    at require (internal/module.js:20:19)
    at bindings (/home/autocular/Documents/Tracker-3/device/node_modules/bindings/bindings.js:76:44)
    at Object.<anonymous> (/home/autocular/Documents/Tracker-3/device/node_modules/serialport/lib/bindings.js:3:35)
    at Module._compile (module.js:570:32)
```
This seems like a few similar issues I've seen filed in this repo for Electron or a RPi but there don't seem to be any other existing issues for X86_64 Ubuntu. 

Any help would be appreciated!

fivdi, 2016-10-17:
See https://github.com/EmergingTechnologyAdvisors/node-serialport/issues/845#issuecomment-226918422

reconbot, 2016-10-30:
Yeah, this wont have anything to do with the platform but with the version of node that installed serialport and the version that's running the code. They need to be the same or you'll get this error. 

I'm going to close this issue for now but if you'd like to continue with it feel free to comment and we'll reopen.

closed 2016-10-30 by reconbot

Arrow7000, 2016-10-31:
Thanks for replying! As it turned out this error happened because I used Yarn to install these packages, and since I use nvm there must've been some kind of inconsistency between Yarn and Node. 

Either way, I've gone back to using the npm client for now, which does work.

reconbot, 2016-10-31:
Yarn ¯\_(ツ)_/¯