serialport/node-serialport #966: Failed to distribute electron app for Windows in Mac dev environment

wyudong, 2016-10-17:
SerialPort version: 4.0.3
NodeJS Version: 5.12.0
Operating System and Hardware: Mac OS
Are you using Electron/Atom/NwJS?: Electron (1.4.3) and Atom

I'm developing an electron app and using node-serialport in it. When I distribute the app for Mac, node-serialport works and everything goes well. However when to distribute the app for Windows using the same source code, serialport fails to work:

> Uncaught Error: %1 is not a valid Win32 application.

I distribute the app using: `electron-packager . --platform=win32 --overwrite`

I think if I can distribute the app on Windows, it can run on Windows as well. So I tried to set the environment on Windows 10 and read the instruments that I could find, including issue #538 Tracking Electron support. But I couldn't even build the app.

Before spending more time on Windows, I want to know if distribute app on Windows is necessary for app running on Windows? Is there any latest working instrument for distributing electron app which uses node-serialport?

jacobq, 2016-10-19:
I have run into a very similar problem with my Electron application. For me, the only reliable solution was to build `serialport` on a Windows computer (note: needed to have Python 2.x installed for some reason) with (from `node_modules/serialport`) using `node-gyp rebuild --target=1.4.3 --arch=x64 --dist-url=https://atom.io/download/atom-shell` (may need to first run `npm install -g node-gyp`) then replace the serialport folder in the builds with that one.

reconbot, 2016-11-02:
I've added some more instructions to get windows `node-gyp` support to the readme. I've actually found it easier to get `electron-builder` and `wine` setup on my mac than getting visual studio on my pc but both will work.

Once node-gyp has what it needs you can rebuild for electron without issue. The reason you don't have to "rebuild" for nodejs is we do it for you and make the binaries available for download.

`electron-rebuild` needs `node-gyp` which needs python 2x but it will figure out the right `dist-url` and do all the heavy lifting for you.

closed 2017-01-21 by reconbot