serialport/node-serialport #969: v5.0.0-beta1 broken on Windows; `TypeError: undefined is not a function`

jacobq, 2016-10-19:
SerialPort version: 5.0.0-beta1
NodeJS Version: 6.9.0
Operating System and Hardware: Windows 10 Pro 64-bit, Lenovo W520 laptop
Are you using Electron/Atom/NwJS?: No (not to demonstrate the bug)

```
$ node -v && npm -v && python --version
v6.9.0
3.8.6
Python 2.7.11
```

### Summary of Problem
In windows, when data is received on a serial port the `data` callback never fires, and even if there is no handler assigned for the `data` event appears in the output regardless:
```
undefined:0

TypeError: undefined is not a function
```

### Expected Behavior
No error should be caused just by receiving data and all `data` callbacks should fire. This seems to work in version 4.0.3. 

### Actual Behavior
Error occurs and callback does not fire, as described in summary above.

### Steps and Code to Reproduce the Issue
Open a port and send data to it (i.e. externally).
Simple demo code available here:
https://github.com/jacobq/serialport-5.0.0-beta1-bug-demo

reconbot, 2016-10-21:
Thanks @jacobq I've got the fix pending in #976 sorry about that

jacobq, 2016-10-21:
That's what I get for using a beta build, haha. Thanks for fixing. Will there be another beta release that includes this or should I grab the latest commits and try it now to see if it works before 5.0.0 ships?

closed 2016-10-21 by reconbot

reconbot, 2016-10-25:
If you could test this before it lands it would help a lot, I don't have
access to the windows machines I was using and our physical windows ci only
runs once a night and isn't integrated.

PR #976 (2016-10-21): put the dataCallback back for windows reads
closes #969