serialport/node-serialport #975: Restore a versioned binary path to aid in distribution and bug reports

reconbot, 2016-10-20:
This is primarily for Electron as it's the classic "I have the wrong version of the binary installed" and "I want to distribute multiple versions of the binary" problem child. This would also help track down a few instances of weirdness in bug reports.

The crux to the issue is that the binary isn't overwritten by default by npm, electron-rebuild (pr's have been merged but not yet released), electron-prebuilt, and who knows what else when the target node version is changed. (via nvm, or targeting electron)

MitchBradley, 2020-01-04:
Any progress on this front?  serialport binary version mismatch problems are the #1 support hassle for the cncjs project.  If I could include a bunch of different precompiled serialport binaries for a few different nodejs versions on a few common platforms, I could make a sleek installation package that Just Works.

MitchBradley, 2020-01-04:
It seems that [prebuildify ](https://github.com/prebuild/prebuildify) might be a good solution.  It lets you package binaries for different architectures X node ABIs X electron/node, to be selected dynamically.  The "node ABI" dimension will collapse when N-ABI becomes the norm.

reconbot, 2020-01-12:
Last week I put about 4 or 5 hours into it with no avail. I'll check out prebuildify.

reconbot, 2020-01-13:
Prebuildify is pretty rad, I'm looking into how to do a cross platform publish but I like this approach a lot and have wanted it for years. ❤️

MitchBradley, 2020-01-14:
Thanks for working on it.  I did a test where I had it compile against a lot of different node versions for one platform, and that worked well.   I haven't tried other platforms yet, and I haven't tried to modify serialport to use the result.  I presume that you can do that easily given the recipe in prebuildify.

reconbot, 2020-01-17:
https://github.com/prebuild/prebuildify/issues/35