serialport/node-serialport #982: Build fails on arm device when installing globally as root

pghalliday, 2016-10-27:
SerialPort version: 4.0.3
NodeJS Version: 6.9.1
Operating System and Hardware: Ubuntu 16.04.1
Are you using Electron/Atom/NwJS?: No

(Run `node -v && npm -v && python --version` to get version numbers)

v6.9.1
3.10.8
Python 2.7.12

### Summary of Problem

I'm trying to install serialport globally while logged in as root and get the following failure

```
root@orangepilite:~# npm install -g serialport
/usr/bin/serialport-list -> /usr/lib/node_modules/serialport/bin/serialport-list.js
/usr/bin/serialport-term -> /usr/lib/node_modules/serialport/bin/serialport-terminal.js

> serialport@4.0.3 install /usr/lib/node_modules/serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download: https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.3/serialport-v4.0.3-node-v48-linux-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.3 and node@6.9.1 (node-v48 ABI) (falling back to source compile with node-gyp)
gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/6.9.1"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/usr/lib/node_modules/serialport/.node-gyp"
make: Entering directory '/usr/lib/node_modules/serialport/build'
make: *** No rule to make target '../.node-gyp/6.9.1/include/node/common.gypi', needed by 'Makefile'.  Stop.
make: Leaving directory '/usr/lib/node_modules/serialport/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Linux 3.4.112-sun8i
gyp ERR! command "/usr/bin/nodejs" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/usr/lib/node_modules/serialport/build/Release/serialport.node" "--module_name=serialport" "--module_path=/usr/lib/node_modules/serialport/build/Release"
gyp ERR! cwd /usr/lib/node_modules/serialport
gyp ERR! node -v v6.9.1
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute '/usr/bin/nodejs /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/usr/lib/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/usr/lib/node_modules/serialport/build/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:877:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
node-pre-gyp ERR! System Linux 3.4.112-sun8i
node-pre-gyp ERR! command "/usr/bin/nodejs" "/usr/lib/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /usr/lib/node_modules/serialport
node-pre-gyp ERR! node -v v6.9.1
node-pre-gyp ERR! node-pre-gyp -v v0.6.30
node-pre-gyp ERR! not ok
Failed to execute '/usr/bin/nodejs /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/usr/lib/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/usr/lib/node_modules/serialport/build/Release' (1)
npm ERR! Linux 3.4.112-sun8i
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install" "-g" "serialport"
npm ERR! node v6.9.1
npm ERR! npm  v3.10.8
npm ERR! code ELIFECYCLE

npm ERR! serialport@4.0.3 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@4.0.3 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /root/npm-debug.log
```

### Steps and Code to Reproduce the Issue

```
sudo su -
npm install -g serialport
```

Note that the build succeeds if I run `sudo npm install -g serialport` as a different user. Also `npm install serialport` succeeds when logged in as root or another user

reconbot, 2016-10-27:
While I don't recommend installing as root, npm requires you to use a flag called "unsafe permissions" if you're going to do so.

`npm install serialport --unsafe-perm`

https://github.com/EmergingTechnologyAdvisors/node-serialport/issues/535#issuecomment-131374653 for more information

closed 2016-10-27 by reconbot

pghalliday, 2016-10-27:
Wow that was fast :)

And it worked, thanks!

I'm running as root because I'm using Chef to automate an install of a NodeJS app that uses serialport. Although I could run Chef with sudo instead of logging on as root I think.

reconbot, 2016-10-27:
That would be about the same, sometimes it's what you got to do. =)

reconbot, 2016-10-30:
Added this to the docs as it comes up enough https://github.com/EmergingTechnologyAdvisors/node-serialport#sudo