serialport/node-serialport #983: Module fail to load with nwjs

BorisBoucher, 2016-10-27:
SerialPort version: 4.0.3 and 5.0.0-beta2
NodeJS Version:6.9.1
Operating System and Hardware: Windows 8 64bits/Intel core i7-4700HQ/8Go RAM
Are you using Electron/Atom/NwJS?: NwJS sdk-0.14.0

(Run `node -v && npm -v && python --version` to get version numbers)

### Summary of Problem
serialport module fail to load when the application is launched from NWJS rather than from node.

The same code used directly in nodejs work fine.

Note: I am using a software serial port loop (free virtual serial port) to connect the application with a simulator running in another nodejs application.
When both application run under nodejs, the two serial ports open fine and the two apps communicate happily.

### Expected Behavior
serial port module shall load fine in NWJS.

### Actual Behavior
The serialport module fail to load in NWJS with the following message:

```
D:\dev\perso\test_nw\node_modules\bindings\bindings.js:86 Uncaught Error: Une routine dinitialisation dune bibliothèque de liens dynamiques (DLL) a échoué.
\\?\D:\dev\perso\test_nw\node_modules\serialport\build\Release\serialport.node
    at Object.Module._extensions..node (module.js:639:18)
    at Module.load (module.js:511:32)
    at tryModuleLoad (module.js:470:12)
    at Function.Module._load (module.js:462:3)
    at Module.require (module.js:521:17)
    at require (internal/module.js:20:19)
    at bindings (D:\dev\perso\test_nw\node_modules\bindings\bindings.js:79:44)
    at Object.<anonymous> (D:\dev\perso\test_nw\node_modules\serialport\lib\bindings.js:3:35)
    at Module._compile (module.js:594:32)
    at Object.Module._extensions..js (module.js:609:10)
```

### Steps and Code to Reproduce the Issue

Create a simple html page for NWJS and add the following javascript code:
```
require("serialport")
```

Zuzon, 2016-10-28:
serialport works with nwjs pretty fine. You just need to copy `node_modules/serialport` to your app root directory

Zuzon, 2016-10-28:
and don't forget to build serialport with nw-gyp

reconbot, 2016-10-28:
I'm not sure about the first part but definitely the second part.

reconbot, 2016-11-06:
Did it work?

BorisBoucher, 2016-11-07:
No, I was not able to make it work, either with NWJS nor Electron. Please understand that I'm not very experienced with the nodejs/javascript ecosystem.

I tested the following :
 - I recompiled the module (it build with 'node-pre-gyp', don't know if this is the required way to "build with nw-gyp"), 
 - put the DLL in the application directory, 
 - put the  serialport module directory in the application directory

All without success.

Finally, I changed my mind and split my application in two process communicating via Websocket, one NWJS app doing the GUI, and on nodejs app doing the serial port stuff.

math3os, 2016-11-08:
this is how to compile it  on a windows x64 :
`node-pre-gyp rebuild --runtime=node-webkit --target=0.18.3 --target_arch=x64 --target-platform=win32`

reconbot, 2016-11-08:
Nw-gyp basically runs that for you

https://github.com/nwjs/nw-gyp#readme

BorisBoucher, 2016-11-09:
Tested with rebuilding the serialport modulewith node-gyp, finally the module load fine in NWJS.

Thank you for the response.

closed 2016-11-09 by BorisBoucher