serialport/node-serialport #984: ERR! stack Error: `C:\Windows\Microsoft.NET\Framework\v4.0.30319\msbuild.exe` failed with exit code: 1

VINTproYKT, 2016-10-30:
SerialPort version: 4.0.3
NodeJS version: 7.0.0
npm version: 3.8.5
Python version: 2.7.6
Operating System and Hardware: Windows 8.1, Lenovo
Using NW.JS

### Summary of Problem

I don't know what steps I have missed for installation, please help.

Bash log:
```
Родион@VINTproYKT-Notebook MINGW64 /c/repo/MyApp
$ npm install serialport --save

> serialport@4.0.3 install C:\repo\MyApp\node_modules\serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download: https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.3/serialport-v4.0.3-node-v51-win32-x64.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.3 and node@7.0.0 (node-v51 ABI) (falling back to source compile with node-gyp)

C:\repo\MyApp\node_modules\serialport>if not defined npm_config_node_gyp (node "C:\Users\▒▒▒▒▒▒\AppData\Roaming\npm\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" clean )  else (node "" clean )

C:\repo\MyApp\node_modules\serialport>if not defined npm_config_node_gyp (node "C:\Users\▒▒▒▒▒▒\AppData\Roaming\npm\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" configure --fallback-to-build --module=C:\repo\MyApp\node_modules\serialport\build\Release\serialport.node --module_name=serialport --module_path=C:\repo\MyApp\node_modules\serialport\build\Release )  else (node "" configure --fallback-to-build --module=C:\repo\MyApp\node_modules\serialport\build\Release\serialport.node --module_name=serialport --module_path=C:\repo\MyApp\node_modules\serialport\build\Release )

C:\repo\MyApp\node_modules\serialport>if not defined npm_config_node_gyp (node "C:\Users\▒▒▒▒▒▒\AppData\Roaming\npm\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" build --fallback-to-build --module=C:\repo\MyApp\node_modules\serialport\build\Release\serialport.node --module_name=serialport --module_path=C:\repo\MyApp\node_modules\serialport\build\Release )  else (node "" build --fallback-to-build --module=C:\repo\MyApp\node_modules\serialport\build\Release\serialport.node --module_name=serialport --module_path=C:\repo\MyApp\node_modules\serialport\build\Release )
▒▒▒▒஥▒▒▒ ▒஥▒⮢ ▒ ▒⮬ ▒襭▒▒ ▒▒ ▒▒▒▒▒▒. ▒⮡▒ ▒▒▒▒▒▒▒ ▒▒ࠫ▒▒▒쭮▒ ▒▒▒▒஥▒▒▒, ▒▒▒▒▒▒▒ ▒▒ࠬ▒▒▒ "/m".
TRACKER : error TRK0005: ▒▒ 㤠▒▒▒▒ ▒▒▒▒▒㦨▒▒ "CL.exe". ▒▒ 㤠▒▒▒▒ ▒▒▒▒ 㪠▒▒▒▒▒ 䠩▒. [C:\repo\MyApp\node_modules\serialport\build\serialport.vcxproj]

gyp ERR! build error
gyp ERR! stack Error: `C:\Windows\Microsoft.NET\Framework\v4.0.30319\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Users\Родион\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\lib\build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Windows_NT 6.3.9600
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\Родион\\AppData\\Roaming\\npm\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "build" "--fallback-to-build" "--module=C:\\repo\\MyApp\\node_modules\\serialport\\build\\Release\\serialport.node" "--module_name=serialport" "--module_path=C:\\repo\\MyApp\\node_modules\\serialport\\build\\Release"
gyp ERR! cwd C:\repo\MyApp\node_modules\serialport
gyp ERR! node -v v7.0.0
gyp ERR! node-gyp -v v3.3.1
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute 'node-gyp.cmd build --fallback-to-build --module=C:\repo\MyApp\node_modules\serialport\build\Release\serialport.node --module_name=serialport --module_path=C:\repo\MyApp\node_modules\serialport\build\Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (C:\repo\MyApp\node_modules\serialport\node_modules\node-pre-gyp\lib\util\compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:877:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
node-pre-gyp ERR! System Windows_NT 6.3.9600
node-pre-gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\repo\\MyApp\\node_modules\\serialport\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd C:\repo\MyApp\node_modules\serialport
node-pre-gyp ERR! node -v v7.0.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.30
node-pre-gyp ERR! not ok
Failed to execute 'node-gyp.cmd build --fallback-to-build --module=C:\repo\MyApp\node_modules\serialport\build\Release\serialport.node --module_name=serialport --module_path=C:\repo\MyApp\node_modules\serialport\build\Release' (1)
npm WARN Failed to parse json
npm WARN Trailing comma in object at 35:1
npm WARN }
npm WARN ^
npm WARN File: C:\repo\MyApp\package.json
npm WARN MyApp No description
npm WARN MyApp No repository field.
npm WARN MyApp No README data
npm WARN MyApp No license field.
npm ERR! Windows_NT 6.3.9600
npm ERR! argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\Родион\\AppData\\Roaming\\npm\\node_modules\\npm\\bin\\npm-cli.js" "install" "serialport" "--save"
npm ERR! node v7.0.0
npm ERR! npm  v3.8.5
npm ERR! code ELIFECYCLE

npm ERR! serialport@4.0.3 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@4.0.3 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     C:\repo\MyApp\npm-debug.log

```

### Expected Behavior
Normal installation

### Actual Behavior
Errors in build

### Steps and Code to Reproduce the Issue
Installed Python 2.7.6 in `C:/Python27` directory and added it to %PATH%. NodeJS and npm are latest.

reconbot, 2016-10-30:
We don't have node 7 prebuilt binaries yet, you'll need a windows
development environment to build the c++ bindings.

reconbot, 2016-10-30:
With serialport@4.0.4 we have node 7 precompiled binaries

closed 2016-10-30 by reconbot