serialport/node-serialport #988: Read data works on windows but not on ubuntu

leonardacs, 2016-11-03:
SerialPort version: 4.0.1
NodeJS Version: 6.6.0
Operating System and Hardware Platform: Ubuntu 16.04 and Windows 7
Are you using an alternative NodeJS runtime? (eg Electron):

### Summary of Problem
I trying to integrate a serial weighting-machine with nodejs.
To request a weight, i need to send a hex 05 to device, then i receive the weigth as result.
When i run the code in windows 7 using virtual box, the serial comunication works fine.
But, when i run on ubuntu, not work correctly.
In ubuntu, the port.on(data) funciton never is called.
The serial configuration is ok, because when i do a loopback test the return is ok.

reconbot, 2016-11-03:
Do you have any sample code? The only difference might be in line endings of the strings you're sending.

leonardacs, 2016-11-03:
My code is very simple:

`var SerialPort = require('serialport');
var ENQ = new Buffer('05', 'hex');

var port = new SerialPort('/dev/ttyUSB0', {
	baudRate: 9600,
	dataBits: 8,
	stopBits: 1,
	parity: 'none',
});

port.on('open', function() {
  port.write(ENQ, function() {
    console.log('Requesting weight...');   
  });
});

port.on('data', function(buffer) {
	console.log(buffer);	
});`

The only difference between linux code and windows code is the port path. In windows i use 'COM3' and linux '/dev/ttyUSB'.
In windows, the loopback test and data reading works perfectly, but in linux only the loopback.

reconbot, 2016-11-05:
I don't see anything obvious. What's the loopback?

reconbot, 2017-01-07:
I'm going to close this issue due to it's age, but if you'd like to continue with it feel free to comment and we'll reopen.

closed 2017-01-07 by reconbot

erikdung, 2017-01-19:
Hi, any progress here?