serialport/node-serialport #990: Memory Leaks

reconbot, 2016-11-03:
We've solved some c++ memory leaks but we might have some JS leaks. I'm opening this and closing #923 as the major leaks on this issue are plugged and we need a non PR to track it.

- [ ] Can we monitor in our CI for leaks?
- [ ] Do we have any now?

jupe, 2016-11-03:
could you bring that fix (https://github.com/EmergingTechnologyAdvisors/node-serialport/pull/989) also for v4 version with new release ? .. because v5 will break backward compatibility it would be very good to have same fix for v4...

reconbot, 2016-11-03:
absolutely https://github.com/EmergingTechnologyAdvisors/node-serialport/pull/923#issuecomment-258246068

reconbot, 2016-11-03:
`serialport@4.0.5` is out. The osx precompiled binaries are still processing but it's published to npm!

reconbot, 2017-01-21:
I'm closing this as nothing has cropped up since.

closed 2017-01-21 by reconbot

Apollon77, 2017-02-08:
I currently also consider a "free(): invalid pointer" / "munmap_chunk(): invalid pointer" / "free(): invalid next size (fast)" thing (i saw all three of them in multiple tries in multiple node versions)

I have an IR-Read-head on my power-meter and this meter is sending data every 2-5 seconds. I have the problem that on linux (armhf ubuntu xenial). I get SIGSEGV (mainly on node 4) and SIGABRT (node 6) with SerialPort 4.0.7. I triedit on my mac working machine with a "socal virtualserial port construct" but wasn't able to reproduce it ... I will try this in the evening on the exact same linux machine with the simulation to see if it happends there too ... If I'm able I will try to write a minimal-reproduce script.
Maybe you have any ideas already or wishes for my tests?

Should we reopen here or create a new ticket?

reconbot, 2017-02-11:
@Apollon77 please open a new ticket, I'd love to see the tests

Apollon77, 2017-02-11:
I too :-( I was´n't able to simplify ... when I try it always works :-(

Apollon77, 2017-02-13:
Small update: The "real application" crashes with node 4/6 around once per hour. In node 7.5 no crash since yesterday 23:00 ... so seems to be that node 7 (and so node) 8 is the solution then :-)

reconbot, 2017-02-13:
Were you able to narrow it down to serialport and node 4/6?

Apollon77, 2017-02-13:
Not really ... I only know that it happends mostly while reading data, but also (minor) times after the dataset was completely read and reading should be paused.

After analyzing deeper and error was mostly while "freeing" pointers I would more assume garbage collection problem, and then it would mean that it was within the Buffer implementation on node ... But this is just an assumption ... maybe proofed by the fact that it is gone in node 7.5 without an different serialport version ...

reconbot, 2017-02-13:
Would you be able to share the OS and serialport version? (Stuff from a new
issue template)

Apollon77, 2017-02-13:
Sure, sorry ...

- __SerialPort Version__: 4.0.7
- __NodeJS Version__: latest nodesource 4, 6 and now 7.5.0
- __Operating System__ and __Hardware Platform__: Ubuntu Xenial 16.04 Armbian on Cubietruck armhf