serialport/node-serialport #994: Finish WindowsBinding reading

reconbot, 2016-11-05:
Follow up from #906 which removed the read loop from the windows binding but didn't yet add in a new `read()` function that worked.

munyirik, 2016-11-11:
This looks like it will require a Read method to be added to serialport.cpp. Is that correct?

munyirik, 2016-11-11:
Should reads be queueable?

reconbot, 2016-11-11:
They don't need to be, we're guaranteeing not to call it again until the
callback returns, as long as the binding guarantees to always return at
least one byte of data. (We use epoll on Unix, I forget the method on
Windows)

munyirik, 2016-11-14:
How were you thinking of handling data events? (since AfterOpenSuccess/EIO_WatchPort have been removed).

reconbot, 2016-11-15:
When someone wants data they ask for it, if no data is available read won't return until there is data. The stream mechanics take care of continuously asking for data in streaming mode. No events needed.

reconbot, 2017-01-21:
This is already done thanks to @munyirik =)

closed 2017-01-21 by reconbot