serialport/node-serialport #995: Test Streams Rewrite Performance and Memory consumption

reconbot, 2016-11-05:
Follow up from #906 to ensure we don't have any memory leaks and are near performant as 4x.

jessicaquynh, 2017-01-25:
@reconbot I would love to give this issue a go! Though, I am not sure where to get started ... is there a similar test found in v4x I can base this from? :)

reconbot, 2017-01-29:
There's not a lot. We have some "manual" tests that might be worth looking at. I've used valgrind to find c++ memory leaks before, and the devtools to look for growing heaps.

https://github.com/EmergingTechnologyAdvisors/node-serialport/tree/master/test/manual-testing

I have no well defined view of what this should look like, or examples from other projects. I wonder what's out there. =)

reconbot, 2017-07-06:
The #1212 improves this so much I don't even care to keep this issue open anymore.

closed 2017-07-06 by reconbot