serialport/node-serialport #999: Virtual Serial Ports

reconbot, 2016-11-05:
We need Serial Ports on our CI otherwise they're not doing all they could for us.

- [ ] linux
- [ ] osx
- [ ] windows

Travis ci supports [socat](http://www.dest-unreach.org/socat/doc/socat.html) on linux and osx I can install it and setup a "virtual Arduino" if someone can can figure a command to set up a pair of virtual ports.

Windows is a bit harder, there are a bunch of emulators but none of them look scriptable. 
- [freevirtualserialports](https://freevirtualserialports.com)
- [com0com](http://com0com.cvs.sourceforge.net/) seems like a good bet, open source
- [vspdxp](http://www.eltima.com/products/vspdxp/) makes their source available might work with us

thom-nic, 2016-11-10:
I've figured out how to use `socat` for testing on Linux & OSX.  It works on CircleCI if you`apt-get socat` in a prebuild step.

The raw command looks like this:
```
socat -d -d pty,raw,nonblock,echo=0,link=ttyV0 pty,raw,nonblock,echo=0,link=ttyV1
```

Here's a snippet from a mocha test where I've got it all wrapped together:
```javascript
const fs = require('fs');
const os = require('os');
const resolve = require('path').resolve;
const child_process = require('child_process');
const SerialPort = require('serialport');

describe('SerialPort', function() {
  let client;
  let hwWrite, hwRead;
  let socat;
  const MASTER_PORT = resolve('./ttyV0');
  const SLAVE_PORT  = resolve('./ttyV1');

  const SOCAT_EXISTS = child_process.spawnSync('socat', ['-h']).status === 0;

  if ( ! SOCAT_EXISTS ) {
    console.warn("`socat` is not installed, skipping serial client tests...");
    const installCmd = os.type() === 'Darwin' ? 'brew install socat' : 'sudo apt-get install socat';
    console.warn(`Please run \`${installCmd}\` to enable these tests!`);
    return;
  }

  beforeEach(done => {
    socat = child_process.spawn('socat', ['-d','-d',
      `pty,raw,nonblock,echo=0,link=${MASTER_PORT}`,
      `pty,raw,nonblock,echo=0,link=${SLAVE_PORT}`],
      {detached: true, stdio: 'ignore'});
    socat.on('close', code => {
      //console.log('socat exited with code', code);
    }).on('error', done);

    // socat needs time to init the fds.  Wait for the expected ports to appear...
    const waitForPorts = () => {
      try {
        if ( fs.statSync(MASTER_PORT).dev && fs.statSync(SLAVE_PORT).dev ) {
          hwWrite = fs.createWriteStream(MASTER_PORT, {autoClose: false});
          hwRead = fs.createReadStream(MASTER_PORT);
          hwRead.on('error', () => {}); // ignore read errors on linux when client side stream is closed.
          client = new SerialPort(SLAVE_PORT, {autoOpen: false});
          done();
        }
      }
      catch( ex ) { // ENOENT if both fds have not been created yet.
        setTimeout(waitForPorts, 100);
      }
    };
    waitForPorts();
  });

  afterEach(done => {
    const cb = err => {
      if ( hwWrite ) hwWrite.close();
      socat.kill('SIGTERM');
      done(err);
    };

    if ( client && client.isOpen() ) client.close(cb);
    else cb();
  });

  // tests go here.  `client` is an initialized serial client and `hwRead` and `hwWrite` are 
  // streams you can use for the 'other' side of the serial client.
});
```

This manages to cleanup the virtual TTY links that socat creates after each test.  If things aren't created and closed in the right order you end up with spurious stream errors but this has been working well for me.  Can't help with windows though :/

reconbot, 2016-11-11:
I cannot wait to give this a try. I'm in negotiations with a windows serialport product, we'll see!

reconbot, 2016-12-14:
I *finally* got this going on travis. It looks like socat doesn't support everything we test with a physical port (custom baudrate test, and flow control flags so far) but it's much better than nothing!

https://travis-ci.org/EmergingTechnologyAdvisors/node-serialport/jobs/183820609

thom-nic, 2016-12-14:
ah yeah it's not perfect but probably will get you at least *some* of the tests you weren't able to do on CI previously?  Guessing you were using an env flag or something to skip some tests on CI?

reconbot, 2016-12-14:
Yep

reconbot, 2016-12-14:
Yep on all accounts

thom-nic, 2016-12-14:
Cool, in that case it sounds like you can get *most* of the tests running on CI at least!  Glad I could help.

reconbot, 2016-12-24:
I have many more of the tests passing by implementing a way to disable specific tests that aren't supported. (this used to be done by platform, now an environment flag) I'm not launching socat from node (though maybe I should) and I don't have the full behavior of the arduino mimicked yet.

fourfathom, 2017-03-20:
Windows: SerialPort.list does not list Virtual Serial Ports, at least not those created by Franson Virtual Serial Port or Franson GPSgate.  I can manually open a virtual "COM1"(for example):  new SerialPort('COM1'....

DraTeots, 2017-07-13:
The same with com0com created ports. 
[image]

In the device manager com0com ports are in their own section. 

At the same time, C# sees all the ports. That is SerialPort.cs source code (if it would help)

```C#
[ResourceExposure(ResourceScope.Machine)]
        [ResourceConsumption(ResourceScope.Machine)]
        public static string[] GetPortNames() {
            RegistryKey baseKey = null;
            RegistryKey serialKey = null;
            
            String[] portNames = null;
 
            RegistryPermission registryPermission = new RegistryPermission(RegistryPermissionAccess.Read, 
                                    @"HKEY_LOCAL_MACHINE\HARDWARE\DEVICEMAP\SERIALCOMM");                                    
            registryPermission.Assert();
 
            try {
                baseKey = Registry.LocalMachine;
                serialKey = baseKey.OpenSubKey(@"HARDWARE\DEVICEMAP\SERIALCOMM", false);
 
                if (serialKey != null) {
 
                    string[] deviceNames = serialKey.GetValueNames();
                    portNames = new String[deviceNames.Length];
 
                    for (int i=0; i<deviceNames.Length; i++)
                        portNames[i] = (string)serialKey.GetValue(deviceNames[i]);    
                }
            }
            finally {
                if (baseKey != null) 
                    baseKey.Close();
                
                if (serialKey != null) 
                    serialKey.Close();
                
                RegistryPermission.RevertAssert();
            }
 
            // If serialKey didn't exist for some reason
            if (portNames == null) 
                portNames = new String[0];
 
            return portNames;
        }
```

P.S. I'm on windows 10 creators update.

vkuehn, 2017-07-13:
as I am also interested I did some search. Looks like named pipes can be used for that.
Here [a dicussion](https://forums.virtualbox.org/viewtopic.php?f=6&t=63585) about that
Would serialport work with named pipes ?

DraTeots, 2017-07-13:
BTW, I tried reading `@"HKEY_LOCAL_MACHINE\DEVICEMAP\SERIALCOMM"`  (inside electron using npm `windows` package) and indeed there are correct device addresses and port names stored as keys and values.

reconbot, 2017-07-16:
> Would serialport work with named pipes ?

@vkuehn [we append `\\.\` to the com port path](https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/6455ecbb518f3644188a19e823db0910e0abd955/src/serialport_win.cpp#L42-L48) that you pass in when creating a port. Per that discussion if you pass `pipe\COM1` in we'll change it to `\\.\pipe\COM1` and you might be set. I don't really know what windows named pipes are but give it a try and report back!

vkuehn, 2017-07-18:
named pipes exist on all OS's. E.g. on Xnix/mac mkfifo mypipe creates a 0 bytes entry which can be used as named pipe. You could simply use cat/echo or any other pipe afine software like a serial terminal 
My Virtual box setting looks like

[image]

which can be used with a serial terminal connecting to com1 and from the physical host connecting to com6 .I tried on windows with putty and on mac with minicom both did work with VB named pipes
Changing the readline.js from serialport examples to COM6 fails with

serialport:binding:auto-detect loading WindowsBinding +0ms
serialport:main opening path: COM6 +18ms
serialport:bindings open +3ms
serialport:main _read queueing _read for after open +6ms
serialport:main Binding #open had an error Error: Opening COM6: File not found
  at Error (native) +3ms

reconbot, 2017-07-18:
Try `pipe\COM6`?

vkuehn, 2017-07-19:
strangely even the putty stuff doesn't work today. Sorry I need a view days to get again a working windows box

vkuehn, 2017-07-19:
btw on a mac serial port connection looks like that how would I use serialport with that ?
[image]

reconbot, 2017-07-19:
Just put the path in there.

---
Francis Gulotta
wizard@roborooter.com

vkuehn, 2017-07-19:
there minicom works but readline.js from the examples fails with
(node:58800) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 1): Error: Error Operation not supported Cannot lock port

reconbot, 2017-07-19:
That's an open option that you can disable called `lock` https://github.com/EmergingTechnologyAdvisors/node-serialport#module_serialport--SerialPort..openOptions

vkuehn, 2017-07-19:
now I get
(node:60420) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 1): Error: Error: Operation not supported on socket setting custom baud rate of 9600
I think setting a baudRate is default but the pip things doesn't allow that...FYI Minitcom is set to 9600 works that way with virtual box

vkuehn, 2017-07-24:
so I have a Test environment again with a windows Host and a virtual box with a windows 7 SP1 installation.
Again putty on both boxes with the pipe as mentioned above works but serialport 5.0.0-beta8 fails with
serialport:binding:auto-detect loading WindowsBinding +0ms
serialport:main opening path: pipe\COM6 +20ms
serialport:bindings open +2ms
serialport:main _read queueing _read for after open +4ms
serialport:main Binding #open had an error Error: Open (GetCommState): Unknown error code 1
  at Error (native) +2ms

reconbot, 2017-07-29:
I don't really know what to do here, `GetComState` must not be supported for pipes as it has no actually bad rate.

reconbot, 2017-07-29:
Alright, This issue was about adding virtual ports to our CI for better testing but has turned into better supporting virtual ports in general. I think.. that's great actually.

This feels super weird, but I think maybe a "skip setup" option might be useful so people can use virtual ports that don't emulate the buadrates and flow control setup commands.

Craytor, 2017-08-02:
General question - saw you guys mentioning virtual com ports... Does serialports able to recognize and use virtual com ports for data ingestion?

reconbot, 2017-08-02:
Often

---
Francis Gulotta
wizard@roborooter.com

hanfengcan, 2017-12-24:
freevirtualserialports always disconnect when I open a virtual port

fourfathom, 2017-12-24:
The virtual serial ports from Franson (GPSgate, etc.) work fine with the NodeRed serial ports.  I don't think you can get the Franson virtual ports in a toolkit anymore, but I use them in my NavMonPc program, and have used them with NodeRed

reconbot, 2018-09-15:
Virtual serialports do not usually work as of now. I'm Happy to land patches and help debug but it's not something I know how to fix as each port seems different. If anyone is a paid customer of a virtual serialport that doesn't work, please open a support ticket with the vendor who makes it. I am happy to work with the vendors to add support for their products.

closed 2018-09-15 by reconbot