serialport/node-serialport #1003: Electron and node-serialport : Can it detect or list USB devices / ports ?

younesherlock, 2016-11-07:
SerialPort version: **1.4.1**
NodeJS Version: **6.9.1**
Operating System and Hardware Platform: **Windows 8.1 / 64 bits**
Are you using an alternative NodeJS runtime? (eg Electron): **Yes, I am using _Electron_**

### Summary of Problem

I am building a desktop app using Electron and I need to list in the first place all the USB devices connected to the computer.

The code below returns the value 0. (**Number of results : 0**)

```
var serialport = require('serialport');

serialport.list(function (err, results) {
    if (err) {
        throw err;
    }
    console.log("Number of results : " + results.length);
});
```

In PowerShell, I executed the command : 

`[System.IO.Ports.SerialPort]::getportnames()`

And it returned nothing as well.

However, when I take a look at my Device Manager, I find this : 

[image]

What do you suggest ?

reconbot, 2016-11-08:
node-usb we only work with serialports a small subset of usb (or other)
devices.

younesherlock, 2016-11-08:
I tried with **node-usb** in the first place, but there are many issues when I install it, plus I need to work with Zadig. Anyway, so, you mean I don't have a chance listing USB devices connected to my laptop with **node-serialport** ?

reconbot, 2016-11-08:
We can list devices that show up as com ports but it looks like you have a ltp port which uses a different subsystem under the hood. LTP1 vs COM1.

closed 2016-11-11 by reconbot