serialport/node-serialport #1004: Build fails for armv7 nodejs 6.9.1

thom-nic, 2016-11-08:
SerialPort version: 4.0.5
NodeJS Version: 6.9.1 (LTS) installed from NodeSource APT repo.
Operating System and Hardware Platform:
`Linux raspberrypi 4.4.21-v7+ #911 SMP Thu Sep 15 14:22:38 BST 2016 armv7l GNU/Linux`

Apologies, looking through the issue tracker I see you get a lot of these but I haven't seen this related to NaN - does it have to do with using Nodejs v6?

```
$ npm i serialport@4

> serialport@4.0.5 install /home/pi/mgt200-express/node_modules/serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.5/serialport-v4.0.5-node-v48-linux-arm.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.5 and node@6.9.1 (node-v48 ABI) (falling back to source compile with node-gyp) 
make: Entering directory '/home/pi/mgt200-express/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
../src/serialport.cpp: In function 'int getIntFromObject(v8::Local<v8::Object>, std::string)':
../src/serialport.cpp:90:52: warning: 'v8::Local<v8::Int32> v8::Value::ToInt32() const' is deprecated (declared at /home/pi/.node-gyp/6.9.1/include/node/v8.h:8202): Use maybe version [-Wdeprecated-declarations]
   return getValueFromObject(options, key)->ToInt32()->Int32Value();
                                                    ^
../src/serialport.cpp: In function 'double getDoubleFromObject(v8::Local<v8::Object>, std::string)':
../src/serialport.cpp:102:53: warning: 'v8::Local<v8::Number> v8::Value::ToNumber() const' is deprecated (declared at /home/pi/.node-gyp/6.9.1/include/node/v8.h:8166): Use maybe version [-Wdeprecated-declarations]
   return getValueFromObject(options, key)->ToNumber()->NumberValue();
                                                     ^
../src/serialport.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE Open(Nan::NAN_METHOD_ARGS_TYPE)':
../src/serialport.cpp:144:19: error: 'class Nan::Callback' has no member named 'Reset'
   baton->callback.Reset(info[2].As<v8::Function>());
                   ^
../src/serialport.cpp: In function 'void EIO_AfterOpen(uv_work_t*)':
../src/serialport.cpp:174:31: warning: 'v8::Local<v8::Int32> v8::Value::ToInt32() const' is deprecated (declared at /home/pi/.node-gyp/6.9.1/include/node/v8.h:8202): Use maybe version [-Wdeprecated-declarations]
     int fd = argv[1]->ToInt32()->Int32Value();
                               ^
../src/serialport.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE Update(Nan::NAN_METHOD_ARGS_TYPE)':
../src/serialport.cpp:193:29: warning: 'v8::Local<v8::Int32> v8::Value::ToInt32() const' is deprecated (declared at /home/pi/.node-gyp/6.9.1/include/node/v8.h:8202): Use maybe version [-Wdeprecated-declarations]
   int fd = info[0]->ToInt32()->Int32Value();
                             ^
../src/serialport.cpp:217:116: warning: 'v8::Local<v8::Int32> v8::Value::ToInt32() const' is deprecated (declared at /home/pi/.node-gyp/6.9.1/include/node/v8.h:8202): Use maybe version [-Wdeprecated-declarations]
   baton->baudRate = Nan::Get(options, Nan::New<v8::String>("baudRate").ToLocalChecked()).ToLocalChecked()->ToInt32()->Int32Value();
                                                                                                                    ^
../src/serialport.cpp:218:19: error: 'class Nan::Callback' has no member named 'Reset'
   baton->callback.Reset(info[2].As<v8::Function>());
                   ^
../src/serialport.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE Write(Nan::NAN_METHOD_ARGS_TYPE)':
../src/serialport.cpp:252:29: warning: 'v8::Local<v8::Int32> v8::Value::ToInt32() const' is deprecated (declared at /home/pi/.node-gyp/6.9.1/include/node/v8.h:8202): Use maybe version [-Wdeprecated-declarations]
   int fd = info[0]->ToInt32()->Int32Value();
                             ^
../src/serialport.cpp:276:19: error: 'class Nan::Callback' has no member named 'Reset'
   baton->callback.Reset(info[2].As<v8::Function>());
                   ^
../src/serialport.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE Close(Nan::NAN_METHOD_ARGS_TYPE)':
../src/serialport.cpp:369:32: warning: 'v8::Local<v8::Int32> v8::Value::ToInt32() const' is deprecated (declared at /home/pi/.node-gyp/6.9.1/include/node/v8.h:8202): Use maybe version [-Wdeprecated-declarations]
   baton->fd = info[0]->ToInt32()->Int32Value();
                                ^
../src/serialport.cpp:370:19: error: 'class Nan::Callback' has no member named 'Reset'
   baton->callback.Reset(info[1].As<v8::Function>());
                   ^
../src/serialport.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE List(Nan::NAN_METHOD_ARGS_TYPE)':
../src/serialport.cpp:418:19: error: 'class Nan::Callback' has no member named 'Reset'
   baton->callback.Reset(info[0].As<v8::Function>());
                   ^
../src/serialport.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE Flush(Nan::NAN_METHOD_ARGS_TYPE)':
../src/serialport.cpp:480:29: warning: 'v8::Local<v8::Int32> v8::Value::ToInt32() const' is deprecated (declared at /home/pi/.node-gyp/6.9.1/include/node/v8.h:8202): Use maybe version [-Wdeprecated-declarations]
   int fd = info[0]->ToInt32()->Int32Value();
                             ^
../src/serialport.cpp:492:19: error: 'class Nan::Callback' has no member named 'Reset'
   baton->callback.Reset(callback);
                   ^
../src/serialport.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE Set(Nan::NAN_METHOD_ARGS_TYPE)':
../src/serialport.cpp:528:29: warning: 'v8::Local<v8::Int32> v8::Value::ToInt32() const' is deprecated (declared at /home/pi/.node-gyp/6.9.1/include/node/v8.h:8202): Use maybe version [-Wdeprecated-declarations]
[... 56 lines cut ...]
npm ERR! If you do, this is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/mgt200-express/npm-debug.log
pi@raspberrypi:~/mgt200-express $ which nodejs
/usr/bin/nodejs
pi@raspberrypi:~/mgt200-express $ which node
/usr/bin/node
pi@raspberrypi:~/mgt200-express $ node --version
v6.9.1
pi@raspberrypi:~/mgt200-express $ nodejs --version
v6.9.1
pi@raspberrypi:~/mgt200-express $ uname -a
Linux raspberrypi 4.4.21-v7+ #911 SMP Thu Sep 15 14:22:38 BST 2016 armv7l GNU/Linux
pi@raspberrypi:~/mgt200-express $ gcc --version
gcc (Raspbian 4.9.2-10) 4.9.2
```

reconbot, 2016-11-08:
Uninstall the system nodejs via apt-get, it looks like it's getting mixed
up.

thom-nic, 2016-11-08:
There's only one version of node installed: `node` is an alternatives symlink to `nodejs`:

```
pi@raspberrypi:~/mgt200-express $ ls -l `which nodejs`
-rwxr-xr-x 1 root root 19865460 Oct 20 00:05 /usr/bin/nodejs
pi@raspberrypi:~/mgt200-express $ ls -l `which node`
lrwxrwxrwx 1 root root 22 Nov  4 21:08 /usr/bin/node -> /etc/alternatives/node
pi@raspberrypi:~/mgt200-express $ update-alternatives --display node
node - auto mode
  link currently points to /usr/bin/nodejs
/usr/bin/nodejs - priority 50
  slave node.1.gz: /usr/share/man/man1/nodejs.1.gz
Current 'best' version is '/usr/bin/nodejs'.
pi@raspberrypi:~/mgt200-express $ update-alternatives --list node
/usr/bin/nodejs  # <-- means no other versions of nodejs installed... right?
```

jupe, 2016-11-08:
I think I got same failure unfortunately

thom-nic, 2016-11-09:
Also tested on BeagleBone Black/ Debian Jessie.  Same error.  

```
debian@beaglebone:~/mgt200-express$ uname -a
Linux beaglebone 4.4.15-ti-r35 #1 SMP Sat Jul 16 01:28:50 UTC 2016 armv7l GNU/Linux
debian@beaglebone:~/mgt200-express$ gcc --version
gcc (Debian 4.9.2-10) 4.9.2

debian@beaglebone:~/mgt200-express$ nodejs --version
v6.9.1
```

Is it possible that source builds are failing in general but most users don't see it because they get the prebuilt version when they `npm install` it?

thom-nic, 2016-11-10:
More investigation here: I can indeed build some versions of 4.x on the raspberry pi and beaglebone.  Digging in further, stay tuned...

thom-nic, 2016-11-10:
Ok some more info.... I got the whole dependency tree to install cleanly.  I believe this may have to do with how npm is resolving semver dependencies and hoisting the `nan` dependency and reusing it across several modules that depend on it:
```
pi@raspberrypi:~/mgt200-express $ npm ls nan
mgt200-express@1.0.0 /home/pi/mgt200-express
├─┬ bcrypt@0.8.7
│ └── nan@2.3.5 
└─┬ serialport@4.0.1
  └── nan@2.4.0 
```

In a failed install:
```
debian@beaglebone:~/mgt200-express$ grep 'version":' node_modules/nan/package.json
  "version": "2.3.5"
```

On the successful install:
```
pi@raspberrypi:~/mgt200-express $ grep 'version":' node_modules/nan/package.json 
  "version": "2.4.0"
```

**To ~~fix~~workaround this I added `"nan": "2.4.0",` as a direct dependency to my project**, which seems to coerce npm to use that version when building serialport.

Indeed, serialport's 4x branch, package.json specifies `"nan": "^2.3.5",`.  The root bug here is probably that serialport needs to require `nan@^2.4.0` in v4.x.

thom-nic, 2016-11-10:
@reconbot would you like me to submit a PR?

reconbot, 2016-11-10:
Please do! You'll have to make it against the 4x branch instead of master.

reconbot, 2016-11-11:
[Version 4.0.6 is Released](https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/tag/4.0.6) please give it a try and let me know if it fixes this.

thom-nic, 2016-11-15:
Yeah I tried this and everything looks good!  Thanks!

PR #1009 (2016-11-10): package.json: update nan dependency to v2.4.0
Fixes #1004 

Because of the fuzzy match in package.json (`"nan": "^2.3.5",`) a normal install *would* pick up the latest nan 2.x (currently v2.4.0) because semver says it's compatible... However when serialport is a peer dependency of another package that uses nan, e.g. `bcrypt` which depends specifically on 2.3.x, npm chooses to install nan 2.3.5 instead.  

This bug probably also triggers only on more recent versions of nodejs (I was using v6.9.1.) Presumably at some point nan v2.3.5 worked just fine.
commit 150a5e4f (2016-11-10): package.json: update nan dependency to v2.4.0

Fixes #1004
closed 2016-11-15 by thom-nic