serialport/node-serialport #1005: Build fails for HypriotOS armv7 nodejs 6.5.0

goliatone, 2016-11-09:
SerialPort version: 4.0.5
NodeJS Version: v6.5.0 
Operating System and Hardware Platform:
hypriot os Version 1.1.0 #Raspbian GNU/Linux 8 (jessie)

### Summary of Problem
Running `serialport` inside a docker container on a Raspberry Pi- build was working with 4.0.1

### Steps and Code to Reproduce the Issue

```
> serialport@4.0.5 install /usr/src/app/node_modules/raspi-serial/node_modules/serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp info it worked if it ends with ok
node-pre-gyp info using node-pre-gyp@0.6.31
node-pre-gyp info using node@6.5.0 | linux | arm
node-pre-gyp info check checked for "/usr/src/app/node_modules/raspi-serial/node_modules/serialport/build/Release/serialport.node" (not found)
node-pre-gyp http GET https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.5/serialport-v4.0.5-node-v48-linux-arm.tar.gz
node-pre-gyp http 404 https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.5/serialport-v4.0.5-node-v48-linux-arm.tar.gz
node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.5/serialport-v4.0.5-node-v48-linux-arm.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.5 and node@6.5.0 (node-v48 ABI) (falling back to source compile with node-gyp) 
node-pre-gyp http 404 status code downloading tarball https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.5/serialport-v4.0.5-node-v48-linux-arm.tar.gz 
gyp info it worked if it ends with ok
gyp info using node-gyp@3.4.0
gyp info using node@6.5.0 | linux | arm
gyp info ok 
gyp info it worked if it ends with ok
gyp info using node-gyp@3.4.0
gyp info using node@6.5.0 | linux | arm
gyp info spawn /usr/bin/python2
gyp info spawn args [ '/usr/src/app/node_modules/node-gyp/gyp/gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'make',
gyp info spawn args   '-I',
gyp info spawn args   '/usr/src/app/node_modules/raspi-serial/node_modules/serialport/build/config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/usr/src/app/node_modules/node-gyp/addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/root/.node-gyp/6.5.0/include/node/common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=/root/.node-gyp/6.5.0',
gyp info spawn args   '-Dnode_gyp_dir=/usr/src/app/node_modules/node-gyp',
gyp info spawn args   '-Dnode_lib_file=node.lib',
gyp info spawn args   '-Dmodule_root_dir=/usr/src/app/node_modules/raspi-serial/node_modules/serialport',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'build',
gyp info spawn args   '-Goutput_dir=.' ]
gyp info ok 
gyp info it worked if it ends with ok
gyp info using node-gyp@3.4.0
gyp info using node@6.5.0 | linux | arm
gyp info spawn make
gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
make: Entering directory '/usr/src/app/node_modules/raspi-serial/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
../src/serialport.cpp: In function ‘int getIntFromObject(v8::Local<v8::Object>, std::string)’:
../src/serialport.cpp:90:52: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated (declared at /root/.node-gyp/6.5.0/include/node/v8.h:8202): Use maybe version [-Wdeprecated-declarations]
   return getValueFromObject(options, key)->ToInt32()->Int32Value();
                                                    ^
../src/serialport.cpp: In function ‘double getDoubleFromObject(v8::Local<v8::Object>, std::string)’:
../src/serialport.cpp:102:53: warning: ‘v8::Local<v8::Number> v8::Value::ToNumber() const’ is deprecated (declared at /root/.node-gyp/6.5.0/include/node/v8.h:8166): Use maybe version [-Wdeprecated-declarations]
   return getValueFromObject(options, key)->ToNumber()->NumberValue();
                                                     ^
../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Open(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:144:19: error: ‘class Nan::Callback’ has no member named ‘Reset’
[... 257 lines cut ...]
npm WARN notsup Not compatible with your operating system or architecture: fsevents@1.0.15
npm ERR! Linux 4.4.24-hypriotos-v7+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "johnny-five@0.10.0" "raspi-io@6.1.1" "serialport@4.0.1"
npm ERR! node v6.5.0
npm ERR! npm  v3.10.3
npm ERR! code ELIFECYCLE

npm ERR! serialport@4.0.5 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@4.0.5 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /usr/src/app/npm-debug.log
```

goliatone, 2016-11-09:
I can confirm that `4.0.1` works fine

thom-nic, 2016-11-09:
Yup, looks like this is the same issue as #1004.  Good to know that 4.0.1 can be used as a stopgap.

thom-nic, 2016-11-10:
See https://github.com/EmergingTechnologyAdvisors/node-serialport/issues/1004#issuecomment-259720600 for a workaround.

reconbot, 2016-11-11:
[Version 4.0.6 is Released](https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/tag/4.0.6) please give it a try and let me know if it fixes this.

closed 2017-07-08 by reconbot