serialport/node-serialport #1011: Disconnect

Zensey, 2016-11-11:
SerialPort version: <=4.0.6, <=3.1.2
NodeJS Version: 

Operating System: Windows 10, amd64
Hardware :  USB UART adapter ch340 for Arduino

### Summary of Problem
When I manually disconnect USB UART adapter ch340 from hub 
- in NodeJS 4.x 'disconnect' event does not occurs (it does'not exit while loop in EIO_WatchPort)
- in NodeJS 6.x program crashes with unhandledException or with following output:
```
undefined:0                                                
                                                           
                                                           
                                                           
TypeError: undefined is not a function  
```

### Steps and Code to Reproduce the Issue
```
const
    co = require('co'),
    log4js = require('log4js'),
    SerialPort = require('serialport');
var logger = log4js.getLogger('test')

var port = 'COM11';

function* test_v4_exc () {
    var sp = new SerialPort(port, {baudRate: 9600, autoOpen: false})
    yield cb => sp.open(cb)
    sp.on('disconnect', function() {
    	logger.info('disconnect');
    })
    sp.on('data', function(d) {
	 logger.info('data', d);
    });
    logger.info('open ok');
}

process.on('uncaughtException', function (err) {
  logger.error('uncaughtException>', err.stack || err);
  process.exit()
});

co(
test_v4_exc
).then(function (res) {
    process.exit()
}, function (err) {
    logger.error('error>', err.stack);
});
```
run this program with CH340 adapter on Windows and manually disconnect adapter from USB.

### Solution 
For node.js 4.x I have a solution: in addition to ERROR_OPERATION_ABORTED error code there can occure one more: ERROR_ACCESS_DENIED. It depends on driver.

serialport_win.cpp
```
...
      if (data->errorCode != ERROR_IO_PENDING) {
        // Read operation error
        if (data->errorCode == ERROR_OPERATION_ABORTED || data->errorCode == ERROR_ACCESS_DENIED) {
...

        // Read operation error
        data->errorCode = GetLastError();
        if (data->errorCode == ERROR_OPERATION_ABORTED || data->errorCode == ERROR_ACCESS_DENIED) {
          data->disconnected = true;
```

For node.js 6.x 
I don't have any solution, program crashes,

Zensey, 2016-11-11:
Should I make a pull request ? Which branch ?

reconbot, 2016-11-11:
Good question, were replacing read for Windows on 5x right now and many
people will get the 4x update before 5x is released.

reconbot, 2016-11-19:
I'd make the PR against 4x, 5x is still in active development

reconbot, 2017-07-06:
Probably fixed in `serialport@5.0.0-beta6` as we take any error during reads or writes as a disconnect. (And errors during writes as an error event too) Thanks for the PR at the time, sorry we couldn't get it merged. Hopefully it's not needed anymore.

closed 2017-07-21 by reconbot