serialport/node-serialport #1013: Port closed when flushing on Windows

samisaham, 2016-11-12:
SerialPort version: 5.0.0-beta2
NodeJS Version: 6.9.1
Operating System and Hardware Platform: Windows 10 home 64-bit
Are you using an alternative NodeJS runtime? (eg Electron): No

### Summary of Problem
The port is disconnected whenever the port is flushed.

Removing the PURGE_RXABORT flag from EIO_Flush()>PurgeComm() in seriaport_win.cpp seems to remove the symptoms. I suspect the 

### Steps and Code to Reproduce the Issue
```

var SerialPort = require('serialport');
var Readline = SerialPort.parsers.Readline;

var parser = new Readline();

var port = new SerialPort("COM3", {
				baudrate: 9600,
				databits: 8,
				stopbits: 1,
				parity: 'none',
				rtscts: true,
				xon: false,
				xoff: false,
				autoOpen: true
			});

port.pipe(parser);

parser.on('data', console.log);

port.on("close",function(){
	console.log("closed");
});

port.on("disconnect",function(){
	console.log("disconnected");
});

port.on("open",function(){
	console.log("opened");

	console.log("flushing...");
	
	port.flush(function(err){
		console.log("Flush error: ",err);
	});
	
});

/*
Output:

opened
flushing...
disconnected
Flush error:  null
closed
*/

```

reconbot, 2016-11-19:
We should make sure this is properly tested when #994 lands

reconbot, 2017-07-10:
This is tested in #1223

closed 2017-07-10 by reconbot