serialport/node-serialport #1014: Drop support for node < 4

jimmywarting, 2016-11-12:


reconbot, 2016-11-13:
It's on our roadmap for v6. Too many embedded devices still use 10 and 12
and there isn't a huge technical reason to drop it yet.

jimmywarting, 2016-11-13:
Node stopped maintaining 0.10 a few days ago so. I just want to reduce the package dependencies like object-assign won't be needed any longer...

reconbot, 2016-11-13:
Agreed, it's been a lot of work to keep the polyfills small. I'd like to
make the 5x branch available to older devices that have deps that won't run
on node 4+ because I'd love to never have to publish another 4x release.
The BeagleBone is blocking this (@rwaldron has been working with people to
break that obstacle) along with a few other platforms I don't recall off
hand.

closed 2016-11-19 by reconbot