serialport/node-serialport #1021: Angular2 + TS + Node.js = Module not found: Error: Can't resolve 'child_process'

MartinMajewski, 2016-11-27:
SerialPort version: 4.0.6
NodeJS Version: v7.2.0
Operating System and Hardware Platform: macOS - Sierra
Are you using an alternative NodeJS runtime? (eg Electron): No

### Summary of Problem
Serialport compilation fails due to following error:

```
ERROR in ./~/serialport/lib/list-unix.js
Module not found: Error: Can't resolve 'child_process' in '/PATH/TO/PROJECT/node_modules/serialport/lib'
 @ ./~/serialport/lib/list-unix.js 4:19-43
 @ ./~/serialport/lib/bindings.js
 @ ./~/serialport/lib/serialport.js
 @ ./src/app/serial-port-mgr/serial-port-mgr.component.ts
 @ ./src/app/app.module.ts
 @ ./src/app/index.ts
 @ ./src/main.ts
 @ multi main
```

Chrome reports:

```
./~/serialport/lib/list-unix.js
Module not found: Error: Can't resolve 'child_process' in '/Path/To/Project/node_modules/serialport/lib'
resolve 'child_process' in '/Path/To/Project/node_modules/serialport/lib'
  Parsed request is a module
  using description file: /Path/To/Project/node_modules/serialport/package.json (relative path: ./lib)
    Field 'browser' doesn't contain a valid alias configuration
  after using description file: /Path/To/Project/node_modules/serialport/package.json (relative path: ./lib)
    resolve as module
      looking for modules in /Path/To/Project/src
        using description file: /Path/To/Project/package.json (relative path: ./src)
          Field 'browser' doesn't contain a valid alias configuration
        after using description file: /Path/To/Project/package.json (relative path: ./src)
          using description file: /Path/To/Project/package.json (relative path: ./src/child_process)
            as directory
              /Path/To/Project/src/child_process doesn't exist
            no extension
              Field 'browser' doesn't contain a valid alias configuration
              /Path/To/Project/src/child_process doesn't exist
            .ts
              Field 'browser' doesn't contain a valid alias configuration
              /Path/To/Project/src/child_process.ts doesn't exist
            .js
              Field 'browser' doesn't contain a valid alias configuration
              /Path/To/Project/src/child_process.js doesn't exist
      looking for modules in /Path/To/Project/node_modules
        using description file: /Path/To/Project/package.json (relative path: ./node_modules)
          Field 'browser' doesn't contain a valid alias configuration
        after using description file: /Path/To/Project/package.json (relative path: ./node_modules)
          using description file: /Path/To/Project/package.json (relative path: ./node_modules/child_process)
            as directory
              /Path/To/Project/node_modules/child_process doesn't exist
            no extension
              Field 'browser' doesn't contain a valid alias configuration
              /Path/To/Project/node_modules/child_process doesn't exist
            .ts
              Field 'browser' doesn't contain a valid alias configuration
              /Path/To/Project/node_modules/child_process.ts doesn't exist
            .js
              Field 'browser' doesn't contain a valid alias configuration
              /Path/To/Project/node_modules/child_process.js doesn't exist
[/Path/To/Project/src/child_process]
[/Path/To/Project/src/child_process]
[/Path/To/Project/src/child_process.ts]
[/Path/To/Project/src/child_process.js]
[/Path/To/Project/node_modules/child_process]
[/Path/To/Project/node_modules/child_process]
[/Path/To/Project/node_modules/child_process.ts]
[/Path/To/Project/node_modules/child_process.js]
 @ ./~/serialport/lib/list-unix.js 4:19-43
 @ ./~/serialport/lib/bindings.js
 @ ./~/serialport/lib/serialport.js
 @ ./src/app/serial-port-mgr/serial-port-mgr.component.ts
 @ ./src/app/app.module.ts
 @ ./src/app/index.ts
 @ ./src/main.ts
 @ multi main
```

### Steps and Code to Reproduce the Issue
Started fresh Angular-CLI project with Webstorm 2016.3
Installed serialport with npm (npm install serialport --save)
Installed typings with npm (npm install @types/serialport --save)
Imported Serialport to component (import * as SerialPort from 'serialport';)
Tried to list all available port on the console:
```
    SerialPort.list(function (err, ports) {
      ports.forEach(function (port) {
        console.log(port.comName);
      });
    });
```
This code was simply embedded in the ngOnInit Method of the component.

No Lint errors. Autocompletion works.
NPM server does not compile

Please help me figure out how to get this running.

reconbot, 2016-11-27:
Where is this code executing? It doesn't work in a browser.

MartinMajewski, 2016-11-27:
Code is not executed at all. It seems to crash the entire building to that even the app-root selector does not get exchanged with the html template.
First I wanted to run the code inside the components constructor, then on the ngOnInit method... Lastly I put it into a dummy function that gets not called at all... same failure inside the TS 2 JS 2 Server pipeline. As I have very little experience with web development (I'm mainly a desktop app programmer) I have no clue where to look closer for the solution.

MartinMajewski, 2016-11-27:
Okay tried it out with a simple React App

import React, { Component } from 'react';
import './App.css';
import * as SerialPort from 'serialport';

```
class App extends React.Component {
  constructor(props){
    super(props);
  }

  componentdidMount(){
      SerialPort.list(function (err, ports) {
          ports.forEach(function(port) {
              console.log(port.comName);
              console.log(port.pnpId);
              console.log(port.manufacturer);
          });
      });
  }

  render() {
    return (
      <div className="App">

          <h1>Hello React!</h1>

      </div>
    );
  }
}

export default App;
```

As soon as the SerialPort part is included the compilation fails:

[image]

reconbot, 2016-11-27:
You're executing the javascript in the browser, or your bundling the javascript to be run in a browser. 

Either way, the browser doesn't have a `child_process` module, and doesn't have access to the underlying OS to ask if it's got serialports. This would work in an electron app, or in a chrome app with https://github.com/garrows/browser-serialport. But serialport can't be run in a normal browser, there's no web serialport spec yet.

MartinMajewski, 2016-11-27:
So what is serialport good for then? I want to access serialports to communicate with hardware, right? Over a "webpage" that is shown inside a browser. Or have I missed some new ways of interacting with the web? I think I don't get the point right now... *confused

Anyway: this message is not produced by Chrome! It is an error page provided bei a node.js dev-server (started with npm start from Webstorm's terminal). The same error message occurs here:

[image]

reconbot, 2016-11-27:
The serialport access is usually done from a webserver. If you want a web
based application that can talk to a local serialport, electron is the best
way to go. In the future we may end up with a web serialport spec in
browsers but that's a ways off. https://whatwg.github.io/serial/ There's a
lot of security concerns and ux to ratify.

-Francis

---
Francis Gulotta
wizard@roborooter.com

MartinMajewski, 2016-11-27:
Okay, now I understand. Thank you.

I am using electron right now and tried to combine it with browser-serialport, but got only an null ports object.
Then I found this site here
https://github.com/paulolc/electron-quick-start-serialport

The script/install section inside the package.json file combined with this information

https://github.com/electron/electron/blob/master/docs/tutorial/using-native-node-modules.md#modules-that-rely-on-node-pre-gyp

got node-serialport working.

8 hours later and three JS frameworks wiser I finally have it running. *yay

reconbot, 2016-11-27:
Congratulations!

It's a long journey, these things are only "easy" after you've done them many times. 

Electron is also a moving target, things have changed a lot over the past year. The majority of issues opened on this project are about electron. We have a small guide on [`electron-rebuild`](https://github.com/EmergingTechnologyAdvisors/node-serialport#electron) in our readme. (It basically does what the node-gyp command in the quickstart project does.)

Glad you got it working! If you have any hints or tips you'd like to add to our readme, to make it easier for the next person I'd be happy to add them.

closed 2016-11-27 by reconbot