serialport/node-serialport #1023: Baud rate is parsed as NaN (???)

radio-miskovice, 2016-12-03:
SerialPort version: 4.0.6
NodeJS Version: 6.9.1 x64 Windows
Operating System and Hardware Platform: Windows 10
Are you using an alternative NodeJS runtime? (eg Electron): don't think so

### Summary of Problem

node seriaport-terminal -p COM4 -b 9600 
returns "Error 87":

 Error Error: Open (SetCommState): Unknown error code 87
    at Error (native)

(same for any other port or baudrate)

Debugging reveals, that args.baud has "value" NaN

### Steps and Code to Reproduce the Issue

radio-miskovice, 2016-12-03:
Obviously this is caused by buggy command line parser.  I got rid of this bug very simply:
1) line 15: removed the last two parameters **parseInt** and **9600**
2) line 46: replaced **args.baud** by **parseInt( args.baud)**

Now it works perfectly. Please have a look what's the problem with **commander** as it parses any number as NaN (also for the other numerical options).

reconbot, 2016-12-10:
@radio-miskovice oddly enough this works fine if I use `Number` or even a function that calls `parseInt` eg `a => parseInt(a)` Which leads me to track down the issue is that the default value is being passed in as the second param to `parseint` which is the radix. A radix of 9600 produces NaN. Thanks for catching this!

PR #1026 (2016-12-11): Fix baud rate parsing in serialport-terminal
Fixes #1023
commit 61d44cd3 (2016-12-11): Fix baud rate parsing in serialport-terminal

Fixes #1023
commit 530e0a68 (2016-12-11): Fix baud rate parsing in serialport-terminal

Fixes #1023
closed 2016-12-11 by reconbot