serialport/node-serialport #1024: serialport drain does not appear to work as expected on Windows 10

digitalhack, 2016-12-04:
SerialPort version: 4.0.6 or 5.0.0-beta2
NodeJS Version: 6.9.1
Operating System and Hardware Platform: Windows 10
Are you using an alternative NodeJS runtime? (eg Electron): No

### Summary of Problem

If I am reading and understanding the documentation the code below should print out ten lines intx 1 - 10 and then port closed on a device attached to the serial port.  

On the Raspberry Pi it works.  

On Windows 10 it gets three lines  intx 1 - 2 and then nothing else.

If the port.drain statement is removed the code when executed on Windows 10 prints out ten lines intx 1 - 10 and then sits as the port is still open.

It would appear that on Windows 10 drain isn't waiting until all the pending data is being written to the com port.

### Steps and Code to Reproduce the Issue

node file.js where file.js contains the code below.  Code has been tested using both hardware serial ports between two systems as well as on Windows using com0com virtual com port pair.  Realterm was used to receive the output.

serialport was installed and built from source with no change in behaviour.

```
'use strict'

var SerialPort = require('serialport');

var port = new SerialPort('COM12',{baudrate: 9600});

port.on ('open', function () {
  var i;
  for (i=0; i < 11; i++) {
    port.write('intx ' + i.toString() + '\r\n');
  }
  port.drain(function() {
    port.close();
  });
});

port.on ('error', function(error) {
  console.log('error: %s', error);
});

port.on ('data', function(data) {
  console.log("data" + data.toString());
});

port.on('close', function() {
  console.log('port closed');
});
```

reconbot, 2017-07-16:
I spent some time figuring out how to test this. I think it's perfectly reasonable to write code like;

```js
port.write(data);
port.drain(() => {
  port.close()
})
```

However there's no guarantee that the os will have received all the data from the loopback device (an arduino, your virtual serialport pair, etc) by the time drain returns and you call close. Depending on OS buffers, some or none of the data would be received by that point. I'm not really sure how to test this. =D

reconbot, 2017-07-17:
This problem is even harder to work with because of how writes are queued in the streams. How writes are retried when the os buffer is full, and a few other factors. The `drain` event doesn't seem to be fired unless we hit the high water mark either.

Historically the write callback tells us all writes have been submitted to the os, so you'll have to wait for that to finish first. However that's dumb, we can do that in js for you. I'm requiring all bindings to ensure in progress writes complete along with the system level calls to `drain`. This is out of parity with `flush` but it's better so I don't really care. =)

PR #1238 (2017-07-17): [all] Have drain wait for pending JS write operations before calling system drain
This should allow drain to be a lot more useful.

Bonus:
- More logging!
- Bluebird in tests with it's great long stack traces!

closes #1024
closed 2017-07-17 by reconbot