serialport/node-serialport #1025: Error loading serialport module

Jonathan55, 2016-12-06:
SerialPort version: **npm LTS release (v4.0.6)**
NodeJS Version: **v6.9.1**
Operating System and Hardware Platform: **Win 7 x64**
Are you using an alternative NodeJS runtime? (eg Electron): **NW.js v0.18.2 x64 (Node v6.8.1)**

### Summary of Problem

Error loading serialport module. 

### Steps and Code to Reproduce the Issue

According to the NW.js docs (http://docs.nwjs.io/en/latest/For%20Users/Advanced/Use%20Native%20Node%20Modules/) it's possible to use Native Node Modules if the Node.js and NW.js have the same node version and architecture, which is the case.

It only require to change '{npm-path}\node_modules\node-gyp\src\win_delay_load_hook.cc' file with the one they provide and install node modules with the command `npm install node-module`, and i did that to install serialport on the app directory.

But when i try to do `var SerialPort = require('serialport');` i get:

```
Uncaught Error: The system cannot find message text for message number 0x%1 in the message file for %2.
\\?\C:\app\node_modules\serialport\build\Release\serialport.node
    at Object.Module._extensions..node (module.js:639:18)
    at Module.load (module.js:511:32)
    at tryModuleLoad (module.js:470:12)
    at Function.Module._load (module.js:462:3)
    at Module.require (module.js:521:17)
    at require (internal/module.js:20:19)
    at bindings (C:\app\node_modules\bindings\bindings.js:76:44)
    at Object.<anonymous> (C:\app\node_modules\serialport\lib\bindings.js:3:35)
    at Module._compile (module.js:594:32)
    at Object.Module._extensions..js (module.js:609:10)
```

reconbot, 2016-12-13:
I don't know how electron does it's compiling nor do I know how I know how to debug it. I'd open an issue at nwjs.

The only think I can thin of is that `node-pre-gyp` is downloading you a binary and you're not replacing it. I'd try to install `serialport` without downloading the binary. You can do this by running `npm install serialport --build-from-source`

Jonathan55, 2016-12-13:
Thank you for your reply.

I tried that, but as i don't have python installed, it gave me a bunch of errors.

As i saw this was unqiue error because i didn't find anyone with the same problem i knew this would be hard to solve, so i start looking for alternatives and i found Chrome Apps API.

> https://developer.chrome.com/apps/app_serial

Which is available on NW.js and worked for me.

reconbot, 2016-12-13:
Great!

closed 2017-01-07 by reconbot