serialport/node-serialport #1029: I tried to upgrade serialport but..

klanzi, 2016-12-12:
SerialPort version: 1.7.4
NodeJS Version: 7.2.1
Operating System and Hardware Platform: OSX 10.2.1
Are you using an alternative NodeJS runtime? (eg Electron):n

### Summary of Problem

I want to upgrade my serialprt, but after I run `npm install serialport
this shown up 

`node-pre-gyp ERR! UNCAUGHT EXCEPTION 
node-pre-gyp ERR! stack Error: Cannot find module 'internal/fs'
node-pre-gyp ERR! stack     at Function.Module._resolveFilename (module.js:470:15)
node-pre-gyp ERR! stack     at Function.Module._load (module.js:418:25)
node-pre-gyp ERR! stack     at Module.require (module.js:498:17)
node-pre-gyp ERR! stack     at require (internal/module.js:20:19)
node-pre-gyp ERR! stack     at evalmachine.<anonymous>:17:20
node-pre-gyp ERR! stack     at Object.<anonymous> (/Users/clansiecheng/node_modules/serialport/node_modules/node-pre-gyp/node_modules/tar/node_modules/fstream/node_modules/graceful-fs/fs.js:11:1)
node-pre-gyp ERR! stack     at Module._compile (module.js:571:32)
node-pre-gyp ERR! stack     at Object.Module._extensions..js (module.js:580:10)
node-pre-gyp ERR! stack     at Module.load (module.js:488:32)
node-pre-gyp ERR! stack     at tryModuleLoad (module.js:447:12)
node-pre-gyp ERR! System Darwin 16.1.0
node-pre-gyp ERR! command "/usr/local/bin/node" "/Users/clansiecheng/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /Users/clansiecheng/node_modules/serialport
node-pre-gyp ERR! node -v v7.2.1
node-pre-gyp ERR! node-pre-gyp -v v0.6.7
node-pre-gyp ERR! This is a bug in `node-pre-gyp`.
node-pre-gyp ERR! Try to update node-pre-gyp and file an issue if it does not help:
node-pre-gyp ERR!     <https://github.com/mapbox/node-pre-gyp/issues>
npm ERR! Darwin 16.1.0
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "serialport"
npm ERR! node v7.2.1
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE

npm ERR! serialport@1.7.4 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 7
npm ERR! 
npm ERR! Failed at the serialport@1.7.4 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /Users/clansiecheng/npm-debug.log`

### Steps and Code to Reproduce the Issue

sorry for this dumb question, I'm really new here :(

reconbot, 2016-12-12:
Do you have an old version of serialport in your package.json? If so take
it out and try again.

klanzi, 2016-12-13:
this is from my package.json in serialport folder 

`"name": "serialport",
  "version": "5.0.0-beta2",
  "description": "Node.js package to access serial ports. Welcome your robotic JavaScript overlords. Better yet, program them!",
  "author": {
    "name": "Chris Williams",
    "email": "voodootikigod@gmail.com",
    "url": "http://www.voodootikigod.com"
  },
  "binary": {
    "module_name": "serialport",
    "module_path": "build/{configuration}/",
    "host": "https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/5.0.0-beta2"
  },
  "main": "./lib",
  "repository": {
    "type": "git",
    "url": "git://github.com/EmergingTechnologyAdvisors/node-serialport.git"
  },
  "keywords": [
    "serialport",
    "johnny-five",
    "serial port",
    "hardware",
    "iot",
    "nodebots"
  ],
  "maintainers": [
    {
      "name": "Jacob Rosenthal",
      "email": "jakerosenthal@gmail.com"
    },
    {
      "name": "Chris Williams",
      "email": "voodootikigod@gmail.com"
    },
    {
      "name": "Joe Ferner",
      "email": "joe.ferner@nearinfinity.com"
    },
    {
      "name": "Jay Beavers",
      "email": "jay@hikinghomeschoolers.org"
    },
    {
      "name": "Rob Giseburt",
      "email": "giseburt@gmail.com"
    },
    {
      "name": "Francis Gulotta",
      "email": "wizard@roborooter.com"
    }
  ],
  "dependencies": {
    "bindings": "1.2.1",
    "buffer-indexof": "^1.1.0",
    "commander": "^2.9.0",
    "debug": "^2.3.2",
    "lie": "^3.1.0",
    "nan": "^2.4.0",
    "node-pre-gyp": "^0.6.31",
    "object.assign": "^4.0.3",
    "readable-stream": "^2.2.2"
  },
  "devDependencies": {
    "chai": "^3.5.0",
    "chai-subset": "^1.2.2",
    "eslint": "^3.10.0",
    "eslint-config-standard": "^6.2.0",
    "eslint-plugin-promise": "^3.0.0",
    "eslint-plugin-standard": "^2.0.1",
    "istanbul": "^0.4.4",
    "jsdoc-to-markdown": "^2.0.1",
    "mocha": "^3.1.1",
    "node-pre-gyp-github": "^1.1.2",
    "proxyquire": "^1.7.10",
    "sinon": "^1.17.3"
  },
  "bundledDependencies": [
    "node-pre-gyp"
  ],
  "engines": {
    "node": ">= 0.10.0"
  },
  "bin": {
    "serialport-list": "./bin/serialport-list.js",
    "serialport-term": "./bin/serialport-terminal.js"
  },
  "license": "MIT",
  "scripts": {
    "arduino-test": "TEST_PORT=$(./bin/find-arduino.js) npm test",
    "docs": "jsdoc2md --no-cache -t .docs/README.hbs --partial .docs/sig-name.hbs --partial .docs/sig-link.hbs --partial .docs/edit-warning.hbs -r table --separators --name-format lib/* > README.md",
    "docs:diff": "jsdoc2md --no-cache -t .docs/README.hbs --partial .docs/sig-name.hbs --partial .docs/sig-link.hbs --partial .docs/edit-warning.hbs -r table --separators --name-format lib/* | diff -q README.md - || (echo 'Docs out of date, run `npm run docs` and commit the new README.md' && false)",
    "install": "node-pre-gyp install --fallback-to-build",
    "lint": "eslint *.js lib/**/*.js test/**/*.js bin/**/*.js examples/**/*.js",
    "rebuild-all": "npm rebuild && node-gyp rebuild",
    "stress": "mocha --no-timeouts test/arduinoTest/stress.js",
    "test": "istanbul cover ./node_modules/mocha/bin/_mocha",
    "valgrind": "TEST_PORT=$(./bin/find-arduino.js) valgrind --leak-check=full node test/manual-testing/memory-binding.js",
    "debugger": "TEST_PORT=$(./bin/find-arduino.js) node --inspect --debug-brk test/manual-testing/performance.js"
  },
  "gypfile": true
}
`

reconbot, 2016-12-13:
> @reconbot <https://github.com/reconbot> do you mean the package.json in
> the serialport folder or npm folder? thanks for your reply :)

in your project's root folder

---
Francis Gulotta
wizard@roborooter.com

reconbot, 2016-12-16:
From the package.json you provided it looks like you're using the 5x beta, I'd use something from 4x and 6x isn't functional yet.

closed 2016-12-16 by reconbot