serialport/node-serialport #1033: How should I get hexadecimal ?

kangschampagne, 2016-12-14:
SerialPort version: "^4.0.6"
NodeJS Version: "v6.2.0"
Operating System and Hardware Platform:
Are you using an alternative NodeJS runtime? (eg Electron):

### Summary of Problem
How should I get hexadecimal ?
code is no problems. Now i need get hexadecimal from serialport, so, how should I get hexadecimal? 

if I transform the string to hex ,
 e.g:  data.toString(16)   //=> Error.. 

So, the "options (optional)" need something?

Or ASCII to Hex ?    Its'n it ?

### Steps and Code to Reproduce the Issue
var SerialPort = require('serialport');
var port = new SerialPort('COM3',{
	"baudRate" : 57600,
	"autoOpen" : true
});
port.on('data', function (data) {
	console.log("orl: "+data.toString(16)+"\r\n");  //=> throw new TypeError
});

reconbot, 2016-12-14:
You want the string 'hex' check out the buffer docs for details.

https://nodejs.org/api/buffer.html#buffer_buf_tostring_encoding_start_end

closed 2016-12-14 by reconbot

kangschampagne, 2016-12-16:
OK，thanks//