serialport/node-serialport #1035: unable to install it on Raspberry Pi 3

woeimun, 2016-12-16:
SerialPort version:
NodeJS Version: v0.10.29
Operating System and Hardware Platform: Raspberry Pi 3 with Jessie Raspbian 
Are you using an alternative NodeJS runtime? (eg Electron): no

Unable to install serialport into Raspberry Pi 3 Jessie
Nodejs: v0.10.29
npm: 1.4.21
node-pre-gyp: v0.6.32
node-gyp: v3.4.0

I have tried the following;
sudo npm install serialport
sudo npm install serialport --unsafe-perm
sudo npm install serialport --unsafe-perm --build-rom-source

attached with is the debug log file. Thank you.

[npm-debug.log.zip](https://github.com/EmergingTechnologyAdvisors/node-serialport/files/656310/npm-debug.log.zip)

reconbot, 2016-12-16:
Upgrade your node and npm if possible

woeimun, 2016-12-16:
Hi reconbot,
Thanks for your respond.
Let me try out, and will get back to you.

woeimun, 2016-12-19:
Hi reconbot,
I have upgraded the following;
node -v v4.2.1
npm -v v2.14.7

I have remove node via the "Add/Remove software".
download the latest node:
   wget http://node-arm.herokuapp.com/node_latest_armhf.deb
installing it:
   sudo dpkg -i node_latest_armhf.deb
installing npm:
   sudo apt-get install npm
installing serialport
   sudo su
   npm install serialport --unsafe-perm

finally got it to work.....
Thank you very much.

woeimun

closed 2016-12-19 by woeimun