serialport/node-serialport #1036: Unable to install on Rasperry CM1 (Revolution Pi)

Matele0, 2016-12-16:
SerialPort version: 4.0.7
NodeJS Version: 7.2.1
Operating System and Hardware Platform: Revolution Pi (debian wheezy RT-patch)
Are you using an alternative NodeJS runtime? (eg Electron): no

### Summary of Problem
Installation quit with error
### Steps and Code to Reproduce the Issue
Same Problem with RPi 1 on jessie (non RT-patch)

[npm-debug.zip](https://github.com/EmergingTechnologyAdvisors/node-serialport/files/657942/npm-debug.zip)

Help please :)

Thanks,
Martin

reconbot, 2016-12-16:
Can you paste the output it gave you at the terminal? I'm having trouble
finding any cause for the failure in the log file.

---
Francis Gulotta
wizard@roborooter.com

reconbot, 2016-12-16:
Also what version of gcc do you have?

Matele0, 2016-12-17:
That version hint of gcc did the trick. I updated to gcc 4.8 (before I had 4.6). Installed again with --unsafe-perm and finally I could install it.

Thank you
Martin

closed 2016-12-17 by Matele0

reconbot, 2016-12-17:
Great!