serialport/node-serialport #1037: Serialport error during p5.js install

singingBikeman, 2016-12-16:
SerialPort version: 1.7.4
NodeJS Version: 6.9.2
Operating System and Hardware Platform: Dell Latitude e5430 running windows 10
Are you using an alternative NodeJS runtime? (eg Electron): p5.js

### Summary of Problem
Serialport install failed during installation of p5.js editor
got error: 
 Failed at the serialport@1.7.4 install script 'node-pre-gyp install --fallback-to-build'.

### Steps and Code to Reproduce the Issue
Clone this repo: git clone https://github.com/processing/p5.js-editor
Enter the repo directory and install the development modules: npm install
Install secondary modules: cd public and then npm install  <----------- ERROR HERE
Install gulp.js globally: npm install gulp -g
From the root directory of the repo run gulp: gulp
Start up the app: npm run app

[npm-debug.txt](https://github.com/EmergingTechnologyAdvisors/node-serialport/files/658334/npm-debug.txt)

reconbot, 2016-12-16:
This version of serialport is super old and won't work with node 6

reconbot, 2016-12-16:
I opened an issue over there, it's using a fork of serialport that's a few years old. https://github.com/processing/p5.js-editor/issues/293

closed 2016-12-16 by reconbot

singingBikeman, 2016-12-16:
Thanks for the response. I'm getting a 404 on that link though

reconbot, 2016-12-16:
That's because they just turned off issues!?

https://github.com/processing/p5.js-editor#this-editor-has-been-deprecated-we-will-not-be-making-further-releases-we-are-turning-our-energy-toward-a-web-editor-which-will-be-released-sometime-next-year-for-now-we-recommend-using-editors-like-brackets-atom-sublime-text-or-openprocessing

---
Francis Gulotta
wizard@roborooter.com