serialport/node-serialport #1038: Can't install 4.0.7 on win10

keyiis, 2016-12-18:
SerialPort version:4.0.7
NodeJS Version:6.9
Operating System and Hardware Platform:win10 x64
Are you using an alternative NodeJS runtime? (eg Electron):no

when i run `npm i serialport` get below error:
```
> serialport@4.0.7 install D:\work\nodejs\angular\angular2-electron\node_modules\serialport
> node-pre-gyp install --fallback-to-build

                                                                                                                      node-pre-gyp ERR! Tried to download(undefined): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v48-win32-x64.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@6.9.2 (node-v48 ABI) (falling back to source compile with node-gyp)
gyp ERR! clean error
gyp ERR! stack Error: EPERM: operation not permitted, unlink 'D:\work\nodejs\angular\angular2-electron\node_modules\serialport\build\Release\obj\serialport\vc120.pdb'
gyp ERR! stack     at Error (native)
gyp ERR! System Windows_NT 10.0.14393
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "D:\\work\\nodejs\\angular\\angular2-electron\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "clean"
gyp ERR! cwd D:\work\nodejs\angular\angular2-electron\node_modules\serialport
gyp ERR! node -v v6.9.2
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute 'C:\Program Files\nodejs\node.exe D:\work\nodejs\angular\angular2-electron\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js clean' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (D:\work\nodejs\angular\angular2-electron\node_modules\serialport\node_modules\node-pre-gyp\lib\util\compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:877:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
node-pre-gyp ERR! System Windows_NT 10.0.14393
node-pre-gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "D:\\work\\nodejs\\angular\\angular2-electron\\node_modules\\serialport\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd D:\work\nodejs\angular\angular2-electron\node_modules\serialport
node-pre-gyp ERR! node -v v6.9.2
node-pre-gyp ERR! node-pre-gyp -v v0.6.32
node-pre-gyp ERR! not ok
Failed to execute 'C:\Program Files\nodejs\node.exe D:\work\nodejs\angular\angular2-electron\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js clean' (1)
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@^1.0.0 (node_modules\chokidar\node_modules\fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.0.15: wanted {"os":"darwin","arch":"any"} (current: {"os":"win32","arch":"x64"})
npm WARN enoent ENOENT: no such file or directory, open 'D:\work\nodejs\angular\angular2-electron\node_modules\nslog\package.json'
npm WARN @angular/core@2.3.1 requires a peer of rxjs@5.0.0-rc.4 but none was installed.
npm WARN @angular/http@2.3.1 requires a peer of rxjs@5.0.0-rc.4 but none was installed.
npm WARN @angular/router@3.3.1 requires a peer of rxjs@5.0.0-rc.4 but none was installed.
npm ERR! Windows_NT 10.0.14393
npm ERR! argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "i" "serialport"
npm ERR! node v6.9.2
npm ERR! npm  v3.10.9
npm ERR! code ELIFECYCLE

npm ERR! serialport@4.0.7 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@4.0.7 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     D:\work\nodejs\angular\angular2-electron\npm-debug.log
```

reconbot, 2016-12-21:
The line 
```
node-pre-gyp ERR! Tried to download(undefined): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v48-win32-x64.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@6.9.2 (node-v48 ABI) (falling back to source compile with node-gyp)
```

Is weird because that file is available to download.

It also seems like you might have a permission's error.

```
gyp ERR! stack Error: EPERM: operation not permitted, unlink 'D:\work\nodejs\angular\angular2-electron\node_modules\serialport\build\Release\obj\serialport\vc120.pdb'
```

Means it couldn't delete that file.

Make sure you have everything node-gyp needs, you can find the instructions for windows in the [readme](https://github.com/EmergingTechnologyAdvisors/node-serialport#windows).

keyiis, 2016-12-21:
@reconbot mybe  a permission's error(file locked) or network error(try to download abi),when i open a new cmd windows after close all cmd windows,it excute successed.but not found laws.

reconbot, 2016-12-21:
So it worked?

keyiis, 2016-12-21:
@reconbot Looks like ok,but i havn't change any code.

closed 2016-12-21 by keyiis