serialport/node-serialport #1039: Uncaught Error: Cannot find module "fs" - Electron App

MartinMajewski, 2016-12-19:
SerialPort version: 4.0.7
NodeJS Version: Node: 6.5.0
Operating System and Hardware Platform: macOS Sierra
Are you using an alternative NodeJS runtime? (eg Electron): Electron

### Summary of Problem
Webpack cannot find fs module

```
ERROR in ./~/serialport/lib/serialport.js
Module not found: Error: Cannot resolve module 'fs' in /Users/martinmajewski/Documents/Development/ElectronAp
ps/CapLiper/node_modules/serialport/lib

...

```

It looks up everywhere but with no luck.

SerialPort was rebuild for electron as your documentation states.

### Steps and Code to Reproduce the Issue

Follow your instruction to install serialport for Electron

reconbot, 2016-12-19:
The c++ loading problem looks different so you probably worked around that and got past what we provide instructions for. `fs` is the core nodejs `fs` module.  I have no idea why that wouldn't work.

Are you using webpack or something? Can you share your code?

MartinMajewski, 2016-12-19:
I found the reason for this:

The problem is, that an Electron App is still a server-client architecture. Serialport runs fine on the main thread, that is the "server-side" (node.js). Calling Serialport from the renderer, which is the Angular 2 part, is not possible due to security restrictions. fs is a node.js module, that is simply not available to the render thread / scope.

Browser-Serialport does not cause this error, but it also does not work at all (no port listings, nothing). Plus, Browser-Serialport does not provide TypeScript typings (at least as npm @types/ package), so its not practical for me.

The correct way to use SerialPort inside an Electron App is to call all functions from the main thread. This is every js-module that is required from the main.js that Electron starts as entry point. The interconnection to the renderer goes through Electron's ipcMain tunnel.

At least in theory... I am implementing this approach right now. Will give you feedback, as soon as I know more.

MartinMajewski, 2016-12-22:
If you are interested in seeing node-serialport implemented inside an Electron app, please check out my small sample project here:

https://github.com/MartinMajewski/Electron_Angular2_TypeScriptOnly_NoWebpack

Should run out of the box, as everything is a local dependency (except NPM of course).

reconbot, 2016-12-31:
Cool!

closed 2016-12-31 by reconbot