serialport/node-serialport #1044: kernel panic on osx 10.9.5

gengshenghong, 2016-12-27:
SerialPort version: 4.0.1
NodeJS Version: v6.9.1
Operating System and Hardware Platform: OSX 10.9.5, MacBook Pro
Are you using an alternative NodeJS runtime? (eg Electron): no

### Summary of Problem
It will cause kernel panic (NMIPI for spinlock acquisition timeout) on osx 10.9.5, I've tried different serial ports hardware like CP2012 (with CP2012 driver installed) and Arduino board (no driver needed). All will get crash when open and close serial port again and again. The crash happened randomly in my real code, so I've created this test case to reproduce. It crashed after random times of open/close.
### Steps and Code to Reproduce the Issue
The code as below:
/*
Mac OS X 10.9.5 kernel panic
Panic(CPU 2): NMIPI for spinlock acquisition timeout, spinlock: 0xffffff80323de1a0, spinlock owner: 0xffffff80345e79e0, current_thread: 0xffffff80345e79e0, spinlock_owner_cpu: 0x2
*/

const serialport = require('serialport');

var sp = new serialport('/dev/cu.usbserial-AJ02ZJ02', {
      baudRate: 57600,
	autoOpen: false
    });

var connect = function (port, callback) {
	sp.open(function() {
		callback(true);
	});
};

var disconnect = function (callback) {
    sp.close(function () {
      callback();
    });
};

function testConnect() {
	disconnect(function() {
		setTimeout(function() {
		connect('/dev/cu.usbserial-AJ02ZJ02', function(ret) {
			console.log(ret);
		//	testConnect();       // no crash (but this code is not possible in real user case)
			setTimeout(testConnect, 0);  // crash
		});
		}, 0);
	});
}

testConnect();

Run with: ./node osx_10.9.5_kp.js, it crash after random times (3 to 10 usually) of printing "true" on console. I've even tried to change "setTimeout" argument to 10 seconds and also crash after about 3-10 times! Any idea and how to work around (or fix) it?

Thanks,
Shenghong

gengshenghong, 2016-12-28:
FYI, upgrading my mac book pro to 10.12.2 (Sierra) and it works now.

reconbot, 2017-06-20:
👍

closed 2017-06-20 by reconbot