serialport/node-serialport #1045: [docs] Signal differences in parser configuration for 4.x -> 5.x

JollyWizard, 2016-12-29:
## Summary of issue

The mechanism for using alternative parsers has changed between 4.x and 5.x, and the code examples for each are version specific.  This is not noted in the upgrade guide, nor in the Readme.

I spent over an hour trying to get the examples to work before realizing that the github front-facing documentation was for a different version series than the published NPM version.

I suspect other newcomers will repeat my mistakes until the 5.x series is published or the documentation is amended.

reconbot, 2016-12-30:
Any suggestions on how to amend them?

reconbot, 2016-12-30:
And hey, I'm sorry you got stuck for so long. I have [the warnings at the top](https://github.com/EmergingTechnologyAdvisors/node-serialport#which-version-of-serialport-would-you-like-documentation-for) but I'm not sure what else I can do.

reconbot, 2017-01-03:
I'm going to close this issue but I'm open to improving this situation.

closed 2017-01-03 by reconbot

JollyWizard, 2017-01-03:
No worries.  Saw the notifications, but been AFK for the holiday.  Thanks for being so friendly about it.  It was not my intention to gripe, but I can see how it might come across as testy.

I posted the issue, because the errors I was getting were esoteric, and the only reference I could find was from another amateur user on the Arduino forums (can't find link ATM). Someone suggested the 4.x syntax, and it worked, but I was the only one to figure out that the issue was trying to refer to the wrong documentation.  At that point, I went looking for what I missed, and checked my currently installed node version, and realized what was going on.

I'm open to doing a fork/PR, but I'm not clear on the scope of the changes.  I can investigate myself as time allows, but it'd be easier to just ask the expert. 

Am I correct in understanding that the difference is that the 4.x system used constructor parameters (of what type?), and the 5.x changes use transform stream class instances attached to the object using the normal `stream.pipe(nextStream)` mechanism?  Is there anything else the users should know about the changes?  Other 5.x changes that are releveant?

As for what more could be done, etc...

At the least, I would include a section for `4.x -> 5.x` in the upgrade guide.  I read this early on, and it was the first thing I checked after realizing the version mismatch, but there wasn't a section there so it didn't really do it's job.

Beyond that,  I think part of the reason this tripped me and whoever else up, is that usually, in my experience, the beta sits in a branch, and the master is the current published version.  I can see the advantages to doing it both ways, so maybe it is just a misconception on my (new to the ecosystem) part.  From my side, I've always looked to the repo pages for documentation, since the npm web view doesn't support relative links (hence [unpkg.com](https://unpkg.com/)).  I haven't had another issue like this, so in my mind the repo landing page is the canonical source for documentation on the current version.

Since that's not your workflow, I would suggest:

* In the version/documentation list at the top of the Readme, add a little decorator to indicate the current npm series:  i.e. `is the latest 4.x releases.` += `**current npm version**`
* A little note at the end of any sections that have changed since 4.x.  i.e. `** changed since 4.x**`

reconbot, 2017-01-03:
We should have an upgrade guide by the first actual release. I opened #1046 for that. I like the `Added in: v5.0.0` turns out jsdoc has something for that. =)