serialport/node-serialport #1054: Reconnect Port change is not covered

vkuehn, 2017-01-06:
SerialPort version: 4.0.7 (same on 4.0.6)
NodeJS Version: v6.9.2
Operating System and Hardware Platform: Ubuntu 14.04LTS x64
Are you using an alternative NodeJS runtime? (eg Electron): no

there is no function which just shows ports in Use. E.g. as the arduino IDE does.

so I tried to iterate over the list of ports and and Open each single Port to see if 
port.('open') happens but this actually does open(on linux) a port where no device is connected.

A reconnect of a device can always cause a port switch, is there any clever solution to this ?

vkuehn, 2017-01-07:
thanks a lot !

reconbot, 2017-01-07:
There currently isn't a way we can get a list of "active" ports on linux. We do provide the vendor and [pnpID information](https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/4.0.7/README.md#listing-ports). You can use that to get to try to match your device if it provides the information. If the device doesn't provide that information there's nothing we can do.

Another alternative is to use libudev with the (`udev`)[https://www.npmjs.com/package/udev] package. We're not leveraging it as it's really had to install on some platforms (pretty strait forward on linux) but it's worth a look.

closed 2017-01-07 by reconbot

vkuehn, 2017-01-09:
will look in the pnpID stuff. do you know why portOpen  Events happen on linux withou a device being connected ? maybe it would be good ebough to only try to connect to devices which caused such an event !?