serialport/node-serialport #1062: Assertion `cb_v->IsFunction()' failed

Lange, 2017-01-10:
SerialPort version: 4.0.7
NodeJS Version: 6.9.3
Operating System and Hardware Platform: Windows 10
Are you using an alternative NodeJS runtime? (eg Electron): No

### Summary of Problem
An assertion failed, causing a crash in Node. Looking at the log output here, I don't 100% know that `serialport` is the culprit? The error seems to suggest that it's `ws` where the error occurred, but I don't know how that could be. The crash is also immediately after a `TypeError` thrown by `serialport`, so it seems like the two events are related. A screenshot of the log output of my program at the time of the crash is below, the failed assertion happens in the middle:

[image]

### Steps and Code to Reproduce the Issue
I unfortunately cannot provide a concise repro at this time. The failure takes a very long time (40+ hours) to occur, and the code in question is part of a much larger application. We believe that the initial cause of the error is happening in the Arduino Due that our code is talking to, but the error in the Arduino still shouldn't be causing crashes in Node.

reconbot, 2017-01-10:
Looks like maybe the serialport disappears and the os closes the file descriptor of the port. And maybe "disconnection" isn't happening. As a work around I try to look for a failed write (with the write callback) and immediately close the port. This should improve in 5x. I'm still working on the disconnect semanticist

reconbot, 2017-07-06:
Serialport 5.0.0-beta6 improves this a lot. Give it a go and let me know how it works.

Lange, 2017-07-06:
We've unfortunately changed our featureset such that we are no longer using any serial communications, so I cannot test this. Thank you for working on this though!

reconbot, 2017-07-06:
No worries!

closed 2017-07-06 by reconbot