serialport/node-serialport #1065: serialport does not install with npm

henkbredell, 2017-01-12:
SerialPort version: latest version as per npm
NodeJS Version: 6.9.4
Operating System and Hardware Platform: Linux raspberrypi 4.4.38-v7+ #938 SMP Thu Dec 15 15:22:21 GMT 2016 armv7l GNU/Linux

Are you using an alternative NodeJS runtime? (eg Electron): no

### Summary of Problem
npm is unable to install serialport

### Steps and Code to Reproduce the Issue
pi@raspberrypi:~ $ sudo npm install serialport -g
/usr/bin/serialport-list -> /usr/lib/node_modules/serialport/bin/serialport-list.js
/usr/bin/serialport-term -> /usr/lib/node_modules/serialport/bin/serialport-terminal.js

> serialport@4.0.7 install /usr/lib/node_modules/serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v48-linux-arm.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@6.9.4 (node-v48 ABI) (falling back to source compile with node-gyp) 
gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/6.9.4"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/usr/lib/node_modules/serialport/.node-gyp"
make: Entering directory '/usr/lib/node_modules/serialport/build'
make: *** No rule to make target '../.node-gyp/6.9.4/include/node/common.gypi', needed by 'Makefile'.  Stop.
make: Leaving directory '/usr/lib/node_modules/serialport/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Linux 4.4.38-v7+
gyp ERR! command "/usr/bin/nodejs" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/usr/lib/node_modules/serialport/build/Release/serialport.node" "--module_name=serialport" "--module_path=/usr/lib/node_modules/serialport/build/Release"
gyp ERR! cwd /usr/lib/node_modules/serialport
gyp ERR! node -v v6.9.4
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/bin/nodejs /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/usr/lib/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/usr/lib/node_modules/serialport/build/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:877:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
node-pre-gyp ERR! System Linux 4.4.38-v7+
node-pre-gyp ERR! command "/usr/bin/nodejs" "/usr/lib/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /usr/lib/node_modules/serialport
node-pre-gyp ERR! node -v v6.9.4
node-pre-gyp ERR! node-pre-gyp -v v0.6.32
node-pre-gyp ERR! not ok 
Failed to execute '/usr/bin/nodejs /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/usr/lib/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/usr/lib/node_modules/serialport/build/Release' (1)
npm ERR! Linux 4.4.38-v7+
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install" "serialport" "-g"
npm ERR! node v6.9.4
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE

npm ERR! serialport@4.0.7 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@4.0.7 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/npm-debug.log

thanks in advance

reconbot, 2017-01-12:
Please read the install instructions for root/sudo. And also the
instructions for RPI

henkbredell, 2017-01-12:
Which install instructions are you referring to? 

I managed to find a work-around for the problem by runing the installation of sqlite3 with the --unsafe-perm setting.  Is this a bad idea?
sudo npm install --unsafe-perm -g sqlite3
I guess it provides the necessary permissions to node-pre-gyp to do what it has to in order for the installation to be completed.

reconbot, 2017-01-12:
That's what our docs say to do too. =)

https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/master/README.md#sudo--root

henkbredell, 2017-01-12:
oops - so sorry, silly me.
thanks a lot

closed 2017-01-12 by henkbredell

FrankenApps, 2017-08-16:
I have the same error. But I did `sudo npm install serialport --unsafe-perm --build-from-source`. 
Software and Hardware Platform: Linux 3.12.32+ on Raspberry Pi. Everything else is same as above.
My npm log is attached.  I think it has nothing to do with the `graceful-fs` thing...

[npm-debug.txt](https://github.com/EmergingTechnologyAdvisors/node-serialport/files/1228182/npm-debug.txt)