serialport/node-serialport #1068: MacOS Sierra: port fails to open properly

maxwellhadley, 2017-01-15:
SerialPort version: 4.0.5, 4.0.6, 4.0.7
NodeJS Version: 4.6.1
Operating System and Hardware Platform: acOS Sierra 10.12.2; Apple iMac 21.5inch, late 2009 3.06 GHz Intel core 2 duo
Are you using an alternative NodeJS runtime? (eg Electron): No

### Summary of Problem
USB serial port does not open successfully. No open event is emitted. I'm creating the port using:

`      self.serialport = new SerialPort(self.device, {
            baudrate: 38400,
            parser: self.rfxtrxParser(),
            autoOpen: false
        });
`

then subsequently calling

`
        self.serialport.open();
`

I have a handler installed on the "open" event (and other handlers on most of the other events as well) which never gets called. If I trace the code into serialport.js, I see `SerialPortBinding.open()` being called and returning apparently OK, however the callback function which is passed to it is never called, so no "open" event is ever emitted.

My exact same application code works correctly on serialport version 4.0.4

### Steps and Code to Reproduce the Issue

reconbot, 2017-01-16:
The only differences between 4.0.4 and 4.0.5 were plugging a memory leak when parsing parity options. Nothing changed in the actual open code. You can review the changes here.

https://github.com/EmergingTechnologyAdvisors/node-serialport/compare/4.0.4...4.0.5

Maybe you could try removing and reinstalling the modules?

maxwellhadley, 2017-01-16:
> 4.0.4...4.0.5

I used 'sudo npm uninstall serialport', changed the required version in my package.json, then 'sudo npm install serialport'. It seemed reproducible but I will try on another Mac when I get a moment. The serial device is an RFXtrx433 which uses an FTDI chip, I think the drivers have been updated recently. My code is at https://github.com/rfxcom/node-rfxcom/tree/change-to-serialport-4

Thanks

reconbot, 2017-01-16:
> ","avatar_image_url":"
> github.com/EmergingTechnologyAdvisors/node-serialport/compare/4.0.4...4.0.5\r\n\r\nMaybe

If you use sudo make sure to read the special installation notes in the
readme.
https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/master/README.md#sudo--root

lynndylanhurley, 2017-01-17:
@maxwellhadley - are you getting an error event? I'm wondering if this is related to what I'm seeing in #1070.

maxwellhadley, 2017-01-17:
I have a listener  on the "error" event and that is not getting called either. I don't think it's quite the same problem. Does 4.0.4 work for you?

lynndylanhurley, 2017-01-17:
What happens when you do something like this?

```js
self.serialport = new SerialPort(self.device, { 
  baudrate: 38400, 
  parser: self.rfxtrxParser()
}, function(errors) {
  console.log('errors', errors);
});
```

maxwellhadley, 2017-01-17:
I'll try that - it may be something to do with the `autoOpen` option. Unfortunately it may be a while before I'm in a position to play with the hardware again.

reconbot, 2017-01-21:
Feel free to comment when you can try again. `autoOpen` default to true for the record. I'm going to close this for now.

closed 2017-01-21 by reconbot

maxwellhadley, 2017-04-07:
OK problem identified! Finally got a chance to play with the hardware agin. The node_modules/.bin directory was owned by root, and as serialport was not the only package using this directory, uninstalling serialport (as root) and reinstalling it as a normal user failed, as the symlink to serialport-list.js could not be created. I uninstalled all the package dependencies (as root) so that the .bin directory was deleted, then installed serialport again as a normal user, then jasmine-node as root, and everything seems to work. I'm not sure why jasmine-node has to be installed as root, but apparently it does.