serialport/node-serialport #1069: Read serial data using function and callback

carlosen14, 2017-01-16:
SerialPort version: 4.0.1 
NodeJS Version: 4.2.2    -    4.0.0
Operating System and Hardware Platform: MacOS Sierra 10.12.1    -    Raspberry Pi Jessie w/ Pixel
Are you using an alternative NodeJS runtime? (eg Electron): No

Is there a way to read the serialport input data using a function, not event emitter?. In my case, i only recive data a milliseconds after i send some data, but i can't get it to work if the data is recived in other function ( `port.on('data', callback)`, using event emitter), just to clarify, in this case i am working with some sensors, and i send data to serial and after that i recive the response, and all this is inside a `for` loop, only care about the first recived as callback after i send some data.

Instead of work with eventEmitter and using `port.on('data', callback)`. I am talking about using a simple function with a callback and a timer, example:

`port.read(<timeOut in milliseconds>3000 ,function(err, callback){
    if(err){ return "error :" + err; }
    valueRead = callback;
});
`

I know maybe it doesn't exists, but . May i get some help to do it myself for this specific task?..

reconbot, 2017-01-17:
We have a `read()` function in the 5x beta branch, that returns data if there is any, it doesn't wait however.

If you want some sort of buffered read (eg, data that isn't read gets buffered until you read it). You might want to work pausing into the mix so the buffer doesn't get to big, but that only works on linux and osx for serialport 4x. This doesn't include a timeout mechanism.

```
let buffer = new Buffer(0);
port.on('data', (data) => {
  buffer = buffer.concat(data)
})

function read(cb) {
  if (buffer.length > 0) {
    process.nextTick(() => {
      let data = buffer
      buffer = new Buffer(0)
      cb(data)  
    })
  } else {
    port.once('data', () => read(cb))
  }
}
```

carlosen14, 2017-01-17:
Is it available for Raspberry OS Raspbian?

reconbot, 2017-01-17:
I'm not sure what your asking, but yes it does work there. See the readme for install instructions.

reconbot, 2017-01-20:
I'm going to close this issue for now, but if you'd like to continue with it feel free to comment and we'll reopen.

closed 2017-01-20 by reconbot