serialport/node-serialport #1070: Error: Resource temporarily unavailable Cannot lock port

lynndylanhurley, 2017-01-17:
SerialPort version: **`4.0.7`**
NodeJS Version: **`7.0.4`**
Operating System and Hardware Platform:
* **macOS Sierra `10.12.2`**
* **MacBook Pro (15-inch, Late 2016)**
* **Arduino Due, Arduino Uno**

Are you using an alternative NodeJS runtime? (eg Electron):
I'm using **Electron `v1.4.14`.**

### Summary of Problem

I'm able to list the serial devices, but when I try to connect to them I get the following error:

~~~
Resource temporarily unavailable Cannot lock port
~~~

### Steps and Code to Reproduce the Issue

I have 3 arduinos connected to my machine. I'm using `SerialPort.list` to get their `comName` values, then creating a new `SerialPort` instance using the `comName` as the `id` value shown below. The `comName` is usually something like `/dev/cu.usbmodem14541`.

This is the code I'm using to open the port:

```js
    const port = new SerialPort(id, {
      baudRate: 9600,
      parser: SerialPort.parsers.readline(),
    }, (error) => {
      if (error) {
        emit({ type: CONNECT_DEVICE_ERROR, id, error });
      }

      if (port.isOpen()) {
        emit({ type: CONNECT_DEVICE_OPEN, id });
      }

      console.log({
        id,
        port,
        isOpen: port.isOpen(),
        path: port.path,
        baudRate: port.baudRate,
        binding: port.binding,
      });
    });

    emit({ type: CONNECT_DEVICE_START, id, port });
    port.on('data', data => emit({ type: CONNECT_DEVICE_DATA, id, data }));
    port.on('disconnect', () => emit({ type: DEVICE_DISCONNECTED, id }));
    port.on('close', () => emit({ type: CONNECT_DEVICE_CLOSE, id }));
    port.on('error', error => emit({ type: CONNECT_DEVICE_ERROR, id, error }));
```

Instead of opening the ports, this emits the error shown above.

I am able to connect to the devices using `platformio device monitor`, which uses the same values that `serialport` is using:
* baudRate: **9600**
* parity: **none**
* dataBits: **8**
* stopBits: **1**

The following image shows that `platformio device monitor` is connected:

[image]

lynndylanhurley, 2017-01-17:
**Update**: I tried connecting using the cli interface. I get the same result.

https://gist.github.com/lynndylanhurley/c2edf304a21db760a02120a6fd856491

[image]

lynndylanhurley, 2017-01-17:
Also tried updating to the latest beta (`5.0.0-beta3`). I get the same result.

reconbot, 2017-01-20:
I don't think you can lock a `cu.XXX` port but only the `tty.xxx` ports. Each device should show up twice, one on each subsystem. You can pass `{lock:false}` into your open options and see if that works for you.

lynndylanhurley, 2017-01-29:
@reconbot - for some reason when I run `SerialPort.list` I only see `cu.xxx` entries, not the `tty.xxx` entries.

For example, this is the result of `ls -al /dev | grep usbmodem`:
```
❯ ls -al /dev | grep usbmodem
crw-rw-rw-   1 root        wheel           20,  11 Jan 29 14:31 cu.usbmodem14511
crw-rw-rw-   1 root        wheel           20,   7 Jan 29 14:31 cu.usbmodem14531
crw-rw-rw-   1 root        wheel           20,   9 Jan 29 14:31 cu.usbmodem14541
crw-rw-rw-   1 root        wheel           20,  10 Jan 29 14:31 tty.usbmodem14511
crw-rw-rw-   1 root        wheel           20,   6 Jan 29 14:31 tty.usbmodem14531
crw-rw-rw-   1 root        wheel           20,   8 Jan 29 14:31 tty.usbmodem14541
```

But this is all I see from 'SerialPort.list`:
```json
{
  "list": [
    {
      "comName": "\/dev\/cu.lpss-serial1"
    },
    {
      "comName": "\/dev\/cu.lpss-serial2"
    },
    {
      "comName": "\/dev\/cu.Bluetooth-Incoming-Port"
    },
    {
      "comName": "\/dev\/cu.usbmodem14531",
      "locationId": "0x14530000",
      "manufacturer": "Arduino (www.arduino.org)",
      "productId": "0x003d",
      "serialNumber": "85438333036351F0C1E0",
      "vendorId": "0x2a03"
    },
    {
      "comName": "\/dev\/cu.usbmodem14541",
      "locationId": "0x14540000",
      "manufacturer": "Arduino (www.arduino.org)",
      "productId": "0x003d",
      "serialNumber": "85438333036351110211",
      "vendorId": "0x2a03"
    },
    {
      "comName": "\/dev\/cu.usbmodem14511",
      "locationId": "0x14510000",
      "manufacturer": "Arduino (www.arduino.cc)",
      "productId": "0x0043",
      "serialNumber": "95538333535351601162",
      "vendorId": "0x2341"
    }
  ]
}
```

Can you think of a reason why I can't see the `tty.xxx` devices from `SerialPort.list`?

reconbot, 2017-01-29:
It's what osx returns by default. On linux we return the tty ports, and this isn't an issue on windows (in fact they can only be locked afaict). I suppose this is just an unhappy medium. 

Our OSX port listing code was written a while ago [it can be found here](https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/master/src/serialport_unix.cpp#L604-L654) I'm not very familiar with it myself.

Did opening it with locking turned off work?

reconbot, 2017-01-29:
I updated the title to reflect the situation you figured out. It's close enough to a bug I think it's worth making it easy to reference. I think it would be nice to be able to open ports that `.list` returns on all platforms with default options.

lynndylanhurley, 2017-01-30:
Thanks @reconbot, opening the port without locking worked 🥇 

Should I close this out, or should I leave it open until the new title issue is resolved?

reconbot, 2017-02-07:
I'll make a new issue for it.

closed 2017-02-07 by reconbot