serialport/node-serialport #1073: List returns no devices, ports.length is zero

fdeitelhoff, 2017-01-19:
SerialPort version: 4.0.7
NodeJS Version: 4.4.5
Operating System and Hardware Platform: Windows 10 x64
Are you using an alternative NodeJS runtime? (eg Electron): Electron 1.4.12

### Summary of Problem

I managed to compile serialport for Electron 1.4.12 to solve a DLL initialization error. But the serialport library can't find any ports.

### Steps and Code to Reproduce the Issue

The following code will return no data (empty array, error is null, ports.length is 0:

`serialPort.list(function (err, ports) {
    console.log("Number of ports available: " + ports.length);
    console.log(err);
    console.log(ports);
    ports.forEach(function(port) {
        console.log(port.comName);
        console.log(port.pnpId);
        console.log(port.manufacturer);
    });
});`

I'm not quite sure what I'm missing. There's no error, warning or something else.

reconbot, 2017-01-19:
Do the ports show if you use serialport from the command line?

fdeitelhoff, 2017-01-19:
@reconbot You mean via the serialport-list command? I tried that and get the following error:

`Module not found
\\?\E:\Masterthesis\Simplex-Electron\node_modules\serialport\build\Release\serialport.node
    at Error (native)
    at Object.Module._extensions..node (module.js:434:18)
    at Module.load (module.js:343:32)
    at Function.Module._load (module.js:300:12)
    at Module.require (module.js:353:17)
    at require (internal/module.js:12:17)
    at bindings (E:\Masterthesis\Simplex-Electron\node_modules\bindings\bindings.js:76:44)
    at Object.<anonymous> (E:\Masterthesis\Simplex-Electron\node_modules\serialport\lib\bindings.js:3:35)
    at Module._compile (module.js:409:26)
    at Object.Module._extensions..js (module.js:416:10)
`

I've looked into the first mentioned directory and it's there including the mentioned file. Not sure what's wrong there.

reconbot, 2017-01-19:
it's compiled for electron instead of node. Could you try it in another
directory or installed globally?

fdeitelhoff, 2017-01-19:
Installed it globally and `serialport-list` is returning no output at all. When I try `serialport-list --version` i get   the correct version number 4.0.7, so the command itself is working I think. :(

reconbot, 2017-01-19:
It may not be a serial device but rather a USB device.

reconbot, 2017-01-20:
I'm going to close this as it seems serialport is working correctly.

closed 2017-01-20 by reconbot

fdeitelhoff, 2017-01-20:
@reconbot Yes, you are right. Serialport was not the problem. It's working correctly.

Thank you for your effort and patience!