serialport/node-serialport #1076: Abort Trap: 6 on successive writes

mikehall314, 2017-01-23:
SerialPort version: 5.0.0-beta3
NodeJS Version: 7.4.0
Operating System and Hardware Platform: macOS Sierra. Also observed on Ubuntu 16.04 (ia32)
Are you using an alternative NodeJS runtime? (eg Electron): Both vanilla node and Electron

### Summary of Problem
Two successive writes to a thermal printer causes node to crash with `Abort Trap: 6`

### Steps and Code to Reproduce the Issue
```js
const SerialPort = require("serialport");
const port = new SerialPort("/dev/cu.usbserial", {baudRate: 9600}, _ => {
    port.write("Hello World\n");
    port.write("Hello World\n");
});
```

### Expected result
Printer prints
```
   Hello World
   Hello World
```

### Actual result
Printer prints
```
    Hello World
```
Then node crashes with the error  `Abort Trap: 6`

wilcoxky, 2017-02-21:
SerialPort version: 5.0.0-beta3
NodeJS Version: 6.9.2

Also having this problem code below:
```javascript
const port = new SerialPort('/dev/tty.usbserial-FT9JAO38', {
  baudRate: 115200,
 });

const Readline = SerialPort.parsers.Readline;
const parser = port.pipe(new Readline());

parser.on('data', (data) => {
  console.log(data);
});

port.on('open', () => {
  console.log('Opened Port')
  write();
})

function write() {
  port.write('Hello', function () {
    port.drain();
  });
  port.write('World', function () {
    port.drain();
  });
}
```

Error:
`Abort Trap: 6`

glumb, 2017-05-09:
SerialPort version: 4.07
babel-node 6.11.4

Same issue when writing more than one time.

reconbot, 2017-07-06:
This is probably fixed in `serialport@5.0.0-beta6` I'm going to close this in favor of #1188 which is a duplicate but active issues to this one.

closed 2017-07-06 by reconbot