serialport/node-serialport #1077: Can'

paulmand3l, 2017-01-24:
SerialPort version: >3.1.1
NodeJS Version: 7.4.0
Operating System and Hardware Platform: OSX 10.11.6, USB2Dynamixel (FT232R USB UART)
Are you using an alternative NodeJS runtime? (eg Electron): No

### Summary of Problem

After 3.1.1, node-serialport no longer allows 1Mbps baud rate.

### Steps and Code to Reproduce the Issue

Install version 3.1.1 on OSX 10.11.6, open a serialport on an FT232R USB UART chip at 1,000,000 baud. It outputs serial stream at 1Mbaud.

`var port = new require('serialport').SerialPort('/dev/tty.usbserial-AI03QEHN', {baudRate: 1000000});
port.write(Buffer.from([255,255,253,0,254,9,0,131,65,0,1,0,201,1,18,151]));`

Stream is sent, USB2Dynamixel rx/tx light blinks and LED on attached servo turns on.

Install version 3.1.2 (or later) on the same system, run the same code, no stream is sent, USB2Dynamixel rx/tx light doesn't blink and servo doesn't respond.

I'm pretty sure it has something to do with the serialport_unix.cpp refactor here: https://github.com/EmergingTechnologyAdvisors/node-serialport/commit/0063e13364072cdb8ba78ec2f00bae3ae6e92538, but I don't know enough about node-serial port, c++, or OSX serial drivers to be able to pinpoint the problem.

paulmand3l, 2017-01-24:
Found a comment here that seems relevant

https://groups.google.com/a/chromium.org/forum/#!topic/chromium-reviews/w42egk2qMTg

```
// cfsetispeed and cfsetospeed sometimes work for custom baud rates on OS 
// X but the IOSSIOSPEED ioctl is more reliable but has to be done after 
// the rest of the port parameters are set or else it will be overwritten. 
```

It looks like before the above commit, the ioctl call comes at the end of the setup function, whereas that commit refactors it to the setBaudRate function, which gets called in setup before the rest of the port parameters get set here https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/0063e13364072cdb8ba78ec2f00bae3ae6e92538/src/serialport_unix.cpp#L328.

paulmand3l, 2017-01-24:
And indeed, if you use a standard baud rate when you first initialize the serial port, then immediately update to a non-standard baud rate, it works as advertised.

```
var serialport = require('serialport');
var port = new serialport('/dev/tty.usbserial-AI03QEHN', {baudRate: 115200}, function (err) {
  port.update({baudRate: 1000000});
  port.write(Buffer.from([255,255,253,0,254,9,0,131,65,0,1,0,201,1,18,151]));
};
```

reconbot, 2017-01-24:
Ooo good catch.

reconbot, 2017-07-15:
I can't reproduce this error but I did move the setBaudrate code and I added a test for setting the 1000000 baud rate.

reconbot, 2017-07-15:
Sorry for the long wait on this, the change will go out in `serialport@5`

PR #1235 (2017-07-15): [unix] Move setting up the baudrate to the end of the open() to better support custom baudrates
- Test 1000000 baud

closes #1077
closed 2017-07-15 by reconbot